UbuntuAsahi/ubuntu-asahi #79: M3 support

hadim, 2023-11-26:
It looks like M3 is not fully supported yet (same features as for M1/M2). 

Once supported upstream, will that be available straight on Ubuntu Asahi?

tobhe, 2023-11-26:
Hi @hadim. Once m3 support lands in Asahi it will also be available for Ubuntu. We try to add support for everything Asahi supports as fast as we can but we always wait for an official release first so the reference platform (Arch and soon Fedora) is usually a tiny bit faster in adding new features.

Regarding snaps: yes they are fully supported, we even came up with a way to have working GPU acceleration with our custom mesa version.

closed 2023-11-26 by hadim

tobhe, 2023-11-29:
Hi @hadim. I don't think it will work at all since no one (even in upstream Asahi) has had time to work on M3 yet. The installer should also currently not allow you to. I recommend a bit of patience, I think the Asahi folks are busy preparing their Fedora release and fixing up the last Audio and HDMI bits. Once that is done, people will have time to start with M3 support.

hadim, 2023-11-29:
ok, got it. Thanks for the explanation!

romain130492, 2024-08-20:
hey, any news about this by any chance!? @tobhe 
Thanks

tobhe, 2024-08-20:
Hey @romain130492, sadly no news yet. Latest update from Asahi can be found at https://discussion.fedoraproject.org/t/asahi-on-m3/117919