UbuntuAsahi/ubuntu-asahi #80: Sway freezes immediately

EdvardSire, 2023-12-08:
Just tried to `sudo apt install sway`, logging out, and choosing sway when logging in. 

The screen is completely black with the cursor rendered, but not responsive.

Might not be an asahi problem, just marking it here for reference.

eslerm, 2023-12-08:
Hi o/

I'm using Sway as my daily driver. Nothing special was needed on my setup.

Are there clues in the logs from your last boot?

EdvardSire, 2023-12-08:
I've come a bit further than last time, I've once seen sway open and the statusbar toggling when i press super. But every other attempt (and current best effort) just says XDG_RUNTIME_DIR is not set in the environment. Aborting.

This is in my .bash_profile (i found this snippet somewhere)
```
if test -z "${XDG_RUNTIME_DIR}"; then
    export XDG_RUNTIME_DIR=/run/user/1000/
    if ! test -d "${XDG_RUNTIME_DIR}"; then
        mkdir "${XDG_RUNTIME_DIR}"
        chmod 0700 "${XDG_RUNTIME_DIR}"
    fi
fi
```

I'm launching sway from tty

eslerm, 2023-12-08:
I've been using the default login menu. There's a setting in the lower right corner where you can select between GNOME and sway.

Does that load?

If not, is there anything interesting in ` journalctl -b -1` on reboot?

EdvardSire, 2023-12-08:
I'm trying to understand the journalctl output. Another thing that i don't really understand is that i3 works out of the box (i can choose it in what i think is gdm) and that it looks like sway is able to render the gray background but not doing *anything* else.

eslerm, 2023-12-08:
It might be the config. Could you try resetting it with `cp /etc/sway/config ~/.config/sway/config` and then launch Sway through the GDM3 Login Screen?
[image]

EdvardSire, 2023-12-08:
i did 
`cd /etc/sway && rm config && wget https://raw.githubusercontent.com/swaywm/sway/master/config.in && mv config.in config && cp /etc/sway/config ~/.config/sway/config`
then
[image]
the moment i press enter after writing my password the the screen freezes with a drawn cursor.

EdvardSire, 2023-12-08:
sway -Vd
[image]

EdvardSire, 2023-12-08:
When I'm logged in on GNOME i run `sudo dmesg --follow` open another tty and run `sway -Vd`

i get
```
[  718.038224] rfkill: input handler enabled
[  718.042764] audit: type=1107 audit(1702068652.894:131): pid=948 uid=102 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/login1" interface="org.freedesktop.DBus.Properties" member="PropertiesChanged" name=":1.15" mask="receive" pid=2417 label="snap.firefox.firefox" peer_pid=973 peer_label="unconfined"
                exe="/usr/bin/dbus-daemon" sauid=102 hostname=? addr=? terminal=?'
[  718.094598] audit: type=1107 audit(1702068652.946:132): pid=948 uid=102 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/login1" interface="org.freedesktop.DBus.Properties" member="PropertiesChanged" name=":1.15" mask="receive" pid=2417 label="snap.firefox.firefox" peer_pid=973 peer_label="unconfined"
                exe="/usr/bin/dbus-daemon" sauid=102 hostname=? addr=? terminal=?'
[  720.621371] audit: type=1107 audit(1702068655.474:133): pid=948 uid=102 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/login1" interface="org.freedesktop.login1.Manager" member="SessionNew" name=":1.15" mask="receive" pid=2417 label="snap.firefox.firefox" peer_pid=973 peer_label="unconfined"
                exe="/usr/bin/dbus-daemon" sauid=102 hostname=? addr=? terminal=?'
[  725.668181] evdi: [I] (card2) Opened by Task 962 (seatd) of process 962 (seatd)
[  725.668914] evdi: [I] (card3) Opened by Task 962 (seatd) of process 962 (seatd)
[  725.669633] evdi: [I] (card4) Opened by Task 962 (seatd) of process 962 (seatd)
[  725.669883] evdi: [I] (card5) Opened by Task 962 (seatd) of process 962 (seatd)
[  725.671894] evdi: [I] (card3) Opened by Task 8385 (sway) of process 8385 (sway)
[  725.689737] evdi: [I] (card3) Closed by Task 8385 (sway) of process 8385 (sway)
[  725.689777] evdi: [I] (card3) Opened by Task 8385 (sway) of process 8385 (sway)
[  725.689808] evdi: [I] (card3) Closed by Task 8385 (sway) of process 8385 (sway)
[  725.689855] evdi: [I] (card3) Closed by Task 8385 (sway) of process 8385 (sway)
[  725.690731] evdi: [I] (card4) Opened by Task 8385 (sway) of process 8385 (sway)
[  725.697646] evdi: [I] (card4) Closed by Task 8385 (sway) of process 8385 (sway)
[  725.697679] evdi: [I] (card4) Opened by Task 8385 (sway) of process 8385 (sway)
[  725.697699] evdi: [I] (card4) Closed by Task 8385 (sway) of process 8385 (sway)
[  725.697721] evdi: [I] (card4) Closed by Task 8385 (sway) of process 8385 (sway)
[  725.698600] evdi: [I] (card5) Opened by Task 8385 (sway) of process 8385 (sway)
[  725.705465] evdi: [I] (card5) Closed by Task 8385 (sway) of process 8385 (sway)
[  725.705498] evdi: [I] (card5) Opened by Task 8385 (sway) of process 8385 (sway)
[  725.705519] evdi: [I] (card5) Closed by Task 8385 (sway) of process 8385 (sway)
[  725.705605] evdi: [I] (card5) Closed by Task 8385 (sway) of process 8385 (sway)
[  725.725523] evdi: [I] (card2) Opened by Task 8385 (sway) of process 8385 (sway)
[  725.730658] evdi: [I] (card2) Closed by Task 8385 (sway) of process 8385 (sway)
[  725.730693] evdi: [I] (card2) Opened by Task 8385 (sway) of process 8385 (sway)
[  725.882551] audit: type=1107 audit(1702068660.734:134): pid=948 uid=102 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/login1" interface="org.freedesktop.DBus.Properties" member="PropertiesChanged" name=":1.15" mask="receive" pid=2417 label="snap.firefox.firefox" peer_pid=973 peer_label="unconfined"
                exe="/usr/bin/dbus-daemon" sauid=102 hostname=? addr=? terminal=?'
[  728.439201] rfkill: input handler disabled
[  728.442997] evdi: [I] (card2) Opened by Task 973 (systemd-logind) of process 973 (systemd-logind)
[  728.448011] evdi: [I] (card3) Opened by Task 973 (systemd-logind) of process 973 (systemd-logind)
[  728.450295] evdi: [I] (card4) Opened by Task 973 (systemd-logind) of process 973 (systemd-logind)
[  728.452692] evdi: [I] (card5) Opened by Task 973 (systemd-logind) of process 973 (systemd-logind)
[  728.460174] audit: type=1107 audit(1702068663.310:135): pid=948 uid=102 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/login1" interface="org.freedesktop.DBus.Properties" member="PropertiesChanged" name=":1.15" mask="receive" pid=2417 label="snap.firefox.firefox" peer_pid=973 peer_label="unconfined"
                exe="/usr/bin/dbus-daemon" sauid=102 hostname=? addr=? terminal=?'
[  728.526619] evdi: [I] (card2) Closed by Task 2190 (gnome-shell) of process 2190 (gnome-shell)
[  728.527167] evdi: [I] (card3) Closed by Task 2190 (gnome-shell) of process 2190 (gnome-shell)
[  728.527471] evdi: [I] (card4) Closed by Task 2190 (gnome-shell) of process 2190 (gnome-shell)
[  728.527825] evdi: [I] (card5) Closed by Task 2190 (gnome-shell) of process 2190 (gnome-shell)
[  728.565102] audit: type=1107 audit(1702068663.414:136): pid=948 uid=102 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/login1" interface="org.freedesktop.DBus.Properties" member="PropertiesChanged" name=":1.15" mask="receive" pid=2417 label="snap.firefox [... cut ...] el="unconfined"
                exe="/usr/bin/dbus-daemon" sauid=102 hostname=? addr=? terminal=?'
[  796.247299] evdi: [I] (card2) Opened by Task 9209 (Xwayland) of process 9209 (Xwayland)
[  796.247307] evdi: [I] (card2) Closed by Task 9209 (Xwayland) of process 9209 (Xwayland)
```

only thing i know about evdi is that it's needed for my external DisplayLink dock. And that i have seen swaynag on my external monitor reacting to when i press super (that's my statusbar toggle) but no mouse input.

eslerm, 2023-12-08:
Thanks for the logs.

I have not ran Ubuntu Asahi 22.04 with Sway, just +23.04.

What happens if the dock is not connected and you start Sway?

EdvardSire, 2023-12-08:
If the dock isn't connected and i start sway it
1. If i log in through GDM the screen just hard freezes
2. If i log in using sway -Vd (logging) i get normal looking sway status messages but it just stops at `preparing wayland server initialization`

Looking at the dmesg output my guess was that my DisplayLink driver has polluted /dev/dri:
```
ls /dev/dri/
by-path  card0  card1  card2  card3  card4  card5  renderD128
```

If i force sway to use card0 (i'm completely lost to why it doesn't try this in the first place) with:
`WLR_DRM_DEVICES=/dev/dri/card0 sway` it works!

closed 2024-02-14 by eslerm