UbuntuAsahi/ubuntu-asahi #82: speakersafetyd & speaker support

SamilKaraduman, 2023-12-25:
Does this Ubuntu build lack speaker support at the moment? Speaker support is available in the [Debian builds](https://packages.debian.org/sid/speakersafetyd), and it seems plausible to transfer the required packages to Ubuntu for seamless functionality. Alternatively, could the absence of speaker support be attributed to [this](https://social.treehouse.systems/@AsahiLinux/111607864744814408) specific issue?

tobhe, 2023-12-27:
Speaker support in Debian is in fact not yet fully there yet. We are working closely with Debian to have everything packaged properly. With the lsp-plugin fix and our newest kernel build we have a working test version running. Full speaker support will probably be released soon once we made sure there aren't any problems.

SamilKaraduman, 2023-12-27:
Thank you for the prompt update. It's great to hear that the speaker support is actively being worked on, and the collaboration with Debian is ongoing. I appreciate the clarification on the current status and the efforts being made to ensure proper packaging.

While you're still here, I've been attempting to enable swap space on Ubuntu without much success, and I was wondering if you could provide some guidance. I've followed the general steps, but it seems like there might be something I'm missing or a specific configuration needed for this. This is what I've done so far:

1. Created a swap file with `sudo fallocate -1 2G /swapfile`
2. Set permissions with `sudo chmod 600 /swapfile`
3. Set up the swap area with `sudo mkswap /swapfile ` and `sudo swapon /swapfile`
4. Added the swap file to the /etc/fstab to make it permanent. 

Your insights or any additional steps you can provide would be greatly appreciated. 

Thank you in advance for your assistance!

Samil

SamilKaraduman, 2023-12-30:
Was keeping up with progress on IRC, and seems like speaker support has been added. One thing I have to mention is that when I adjust volume, that volume adjusting feedback sound is played really weirdly, like it has some popping and it just sounds awful. Aside from that, everything else works flawlessly.

closed 2023-12-30 by SamilKaraduman