UbuntuAsahi/ubuntu-asahi #84: Trying to add Finnish language support (keyboard)

TeroKeso, 2023-12-27:
```
sudo gnome-language-selector
[sudo] password for ---: 

[<Package: name:'libreoffice-voikko' architecture='arm64' id:7017>, <Package: name:'language-pack-fi' architecture='arm64' id:3555>, <Package: name:'libreoffice-l10n-fi' architecture='arm64' id:6492>, <Package: name:'language-pack-fi-base' architecture='arm64' id:3556>, <Package: name:'thunderbird-locale-fi' architecture='arm64' id:9920>, <Package: name:'python3-libvoikko' architecture='arm64' id:7018>, <Package: name:'libreoffice-help-fi' architecture='arm64' id:6491>, <Package: name:'libvoikko1' architecture='arm64' id:5026>, <Package: name:'language-pack-gnome-fi-base' architecture='arm64' id:3558>, <Package: name:'libreoffice-help-common' architecture='arm64' id:6468>, <Package: name:'gnome-user-docs-fi' architecture='arm64' id:2624>, <Package: name:'language-pack-gnome-fi' architecture='arm64' id:3557>, <Package: name:'libenchant-2-voikko' architecture='arm64' id:5024>, <Package: name:'voikko-fi' architecture='arm64' id:7019>, <Package: name:'hyphen-fi' architecture='arm64' id:3147>, <Package: name:'libhfstospell11' architecture='arm64' id:2895>]
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 173, in activate_name_owner
    return self.get_name_owner(bus_name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 348, in get_name_owner
    return self.call_blocking(BUS_DAEMON_NAME, BUS_DAEMON_PATH,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 634, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.debian.apt': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/defer/__init__.py", line 487, in _inline_callbacks
    result = gen.send(result)
             ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/aptdaemon/client.py", line 1623, in _run_transaction_helper
    daemon = get_aptdaemon(self.bus)
             ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/aptdaemon/client.py", line 1709, in get_aptdaemon
    return dbus.Interface(bus.get_object("org.debian.apt",
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 237, in get_object
    return self.ProxyObjectClass(self, bus_name, object_path,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 250, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 178, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 273, in start_service_by_name
    return (True, self.call_blocking(BUS_DAEMON_NAME, BUS_DAEMON_PATH,
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 634, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.Spawn.ExecFailed: Failed to execute program org.debian.apt: Permission denied
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 173, in activate_name_owner
    return self.get_name_owner(bus_name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 348, in get_name_owner
    return self.call_blocking(BUS_DAEMON_NAME, BUS_DAEMON_PATH,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 634, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.debian.apt': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/defer/__init__.py", line 487, in _inline_callbacks
    result = gen.send(result)
             ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/aptdaemon/client.py", line 1623, in _run_transaction_helper
    daemon = get_aptdaemon(self.bus)
             ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/aptdaemon/client.py", line 1709, in get_aptdaemon
    return dbus.Interface(bus.get_object("org.debian.apt",
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 237, in get_object
    return self.ProxyObjectClass(self, bus_name, object_path,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 250, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 178, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 273, in start_service_by_name
    return (True, self.call_blocking(BUS_DAEMON_NAME, BUS_DAEMON_PATH,
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 634, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.Spawn.ExecFailed: Failed to execute program org.debian.apt: Permission denied
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 173, in activate_name_owner
    return self.get_name_owner(bus_name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 348, in get_name_owner
    return self.call_blocking(BUS_DAEMON_NAME, BUS_DAEMON_PATH,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 634, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.debian.apt': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/aptdaemon/client.py", line 1594, in on_error
    error.raise_exception()
  File "/usr/lib/python3/dist-packages/defer/__init__.py", line 130, in raise_exception
    raise self.value.with_traceback(self.traceback)
  File "/usr/lib/python3/dist-packages/defer/__init__.py", line 487, in _inline_callbacks
    result = gen.send(result)
             ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/aptdaemon/client.py", line 1623, in _run_transaction_helper
    daemon = get_aptdaemon(self.bus)
             ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/aptdaemon/client.py", line 1709, in get_aptdaemon
    return dbus.Interface(bus.get_object("org.debian.apt",
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 237, in get_object
    return self.ProxyObjectClass(self, bus_name, object_path,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 250, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 178, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 273, in start_service_by_name
    return (True, self.call_blocking(BUS_DAEMON_NAME, BUS_DAEMON_PATH,
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 634, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.Spawn.ExecFailed: Failed to execute program org.debian.apt: Permission denied

```

[image]

mauromarzocca, 2024-01-11:
I resolved using these commands:
`sudo apt-get install aptitude`
`sudo aptitude reinstall dbus`

For my language (Italian) works

closed 2024-01-12 by eslerm

eslerm, 2024-01-12:
Did this issue persist between reboots?

TeroKeso, 2024-01-13:
> Did this issue persist between reboots? Which release were you running?

Running Ubuntu 23.10 / 
_Linux localhost.localdomain 6.5.0-1013-apple-arm #15-Ubuntu SMP PREEMPT_DYNAMIC Sat Nov 18 05:39:03 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux_ 

and tested also with 

_Linux localhost.localdomain 6.6.0-1003-apple-arm #3-Ubuntu SMP PREEMPT_DYNAMIC Wed Dec 27 01:40:02 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux_

with same problems. Did end up to install all language packages by hand, but language support GUI is still crashing.  I will try to fix with @mauromarzocca instructions.