UbuntuAsahi/ubuntu-asahi #86: confused about asahi ubuntu hardware support

parisni, 2024-01-13:
```shell
hostnamectl
  Operating System: Ubuntu 23.10
            Kernel: Linux 6.6.0-1003-apple-arm
      Architecture: arm64
```

This doc maintain the hardware support https://leo3418.github.io/asahi-wiki-build/m2-series-feature-support/
In particular :
```
Kernel release, e.g. 6.0: the feature was incorporated upstream as of this release
linux-asahi (kernel release): the feature is stable, available for use in Fedora Asahi Remix, and should be upstream by the release indicated
linux-asahi: the feature is (mostly) stable and available for use in Fedora Asahi Remix
```

Dicussion here often mention kernels linux-asahi-edge versus linux-asahi.

I am not sure if ubuntu asahi support full hardware as well as fedora/archlinux. Also I cannot find the edge kernel within the apt search.

Any clarification welcome

tobhe, 2024-01-13:
Hi @parisni. We are indeed usually up-to-date and on-par with fedora (Arch Linux actually lags behind nowadays). When new features are introduced it might take us a few weeks to close the gap. At the moment I am not aware of anything missing in Ubuntu.

The -edge kernel used to be a thing but nowadays it is not needed anymore since everything is included in the regular kernel.

I think you are right in that we could advertise our hardware support better here or on our website. Thanks for the feedback!

parisni, 2024-01-13:
> [Tracking image][https://github.com/notifications/beacon/ACIX65UQHVR3BVN2HKY7RLDYOJXAVA5CNFSM6AAAAABBZFOOK6WGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTQVWPSC.gif]

Thanks that helps. There might be some stuff Ubuntu user can setup after installation such as GPU acceleration see https://asahilinux.org/2022/12/gpu-drivers-now-in-asahi-linux/
The tutorial targets only arch users.

So yes few documentation for Ubuntu to either mention everything is enabled by default would make the project more attractive

tobhe, 2024-01-13:
> Thanks that helps. There might be some stuff Ubuntu user can setup after installation such as GPU acceleration see https://asahilinux.org/2022/12/gpu-drivers-now-in-asahi-linux/ The tutorial targets only arch users.

That is because there are no additional steps necessary, it should all just work. There isn't really anything we expect users to set up manually.

I'll see if we can add those infos to a FAQ section on our website.

closed 2024-01-25 by parisni