UbuntuAsahi/ubuntu-asahi #88: After running terminal cmd to run installer

ipillyx, 2024-02-05:
02-05 19:03 root         INFO     Startup
02-05 19:03 root         INFO     Environment:
02-05 19:03 root         INFO       INSTALLER_BASE=https://cdn.asahilinux.org/installer
02-05 19:03 root         INFO       INSTALLER_DATA=https://tobhe.de/ubuntu/installer_data.json
02-05 19:03 root         INFO       REPO_BASE=https://files.tobhe.de/ubuntu
02-05 19:03 root         INFO       IPSW_BASE=None
02-05 19:03 root         INFO       EXPERT=None
02-05 19:03 root         INFO       REPORT=None
02-05 19:03 root         INFO       REPORT_TAG=None
02-05 19:03 root         INFO     Version: v0.7.1
02-05 19:03 root         INFO     MSG: Welcome to the Asahi Linux installer!
02-05 19:03 root         INFO     MSG: This installer will guide you through the process of setting up
02-05 19:03 root         INFO     MSG: Asahi Linux on your Mac.
02-05 19:03 root         INFO     MSG: Please make sure you are familiar with our documentation at:
02-05 19:03 root         INFO     MSG:   [34;1mhttps://alx.sh/w[m[m
02-05 19:03 root         INFO     MSG: Press enter to continue.
02-05 19:03 root         INFO     MSG: Collecting system information...
02-05 19:03 root         ERROR    Exception caught
Traceback (most recent call last):
  File "/private/tmp/asahi-install/main.py", line 1058, in <module>
    InstallerMain(installer_version).main()
  File "/private/tmp/asahi-install/main.py", line 831, in main
    self.sysinfo = system.SystemInfo()
  File "/private/tmp/asahi-install/system.py", line 8, in __init__
    self.fetch()
  File "/private/tmp/asahi-install/system.py", line 18, in fetch
    self.ioservice = plistlib.loads(result.stdout)
  File "/private/tmp/asahi-install/Frameworks/Python.framework/Versions/3.9/lib/python3.9/plistlib.py", line 883, in loads
    return load(fp, fmt=fmt, dict_type=dict_type)
  File "/private/tmp/asahi-install/Frameworks/Python.framework/Versions/3.9/lib/python3.9/plistlib.py", line 875, in load
    return p.parse(fp)
  File "/private/tmp/asahi-install/Frameworks/Python.framework/Versions/3.9/lib/python3.9/plistlib.py", line 177, in parse
    self.parser.ParseFile(fileobj)
xml.parsers.expat.ExpatError: not well-formed (invalid token): line 846509, column 38
02-05 19:03 root         INFO     MSG: If you need to file a bug report, please attach the log file:
02-05 19:03 root         INFO     MSG:   /private/tmp/asahi-install/installer.log

tobhe, 2024-02-05:
Not sure what might be causing the error. Maybe there are network problems on the server side. Could you please try again in a few minutes and let us know if it is still broken? I will meanwhile see if I can reproduce the problem.

eslerm, 2024-02-18:
Closing as this may have been resolved. Please ask to reopen if this continues to affect Ubuntu Asahi and not Asahi Linux.

closed 2024-02-18 by eslerm