UbuntuAsahi/ubuntu-asahi #90: ROS + Ubuntu 22.04

peterdavidfagan, 2024-02-10:
Thanks for all the work that goes into UbuntuAsahi, it's an awesome project. 

I was trying to get a Linux distribution on my MacBook Air (M2 chip) to run with ROS 2. I saw that 22.04 support was recently removed, I wished to ask if there exists an older install script that could be used to build 22.04. It may be sufficient for my use case so wished to ask if it exists. 

(I don't have as much of a requirement for functioning ports etc. more that I can compile code + use GPU drivers for rendering with OpenGL)

ROS releases will eventually move to the next Ubuntu LTS but in the meantime I wished to see if I could get 22.04 running as latest versions of ROS strictly depend on 22.04.

eslerm, 2024-02-11:
Hi @peterdavidfagan o/

There was another recent mention abut ROS on Ubuntu Asahi. Would this work for your use case: https://github.com/UbuntuAsahi/ubuntu-asahi/issues/87#issuecomment-1937813625 ?

peterdavidfagan, 2024-02-12:
Thanks @eslerm, 

I need access to the GPU for rendering with OpenGL and in some cases ML related debugging (when the library supports Apple's metal). Right now Docker only seems to have support for the Nvidia drivers (based on my current understanding) and as a result I have been looking to run ROS software on host with UbuntuAsahi. 

 It seems possible to use 23.10, it just requires building all software dependencies from source which is a bit non-trivial in the case of large ROS repositories as there are a lot of ROS package and system dependencies. I think I found a tool which helps though (rosinstall_generators). 

I can probably build everything from source for now and when ROS moves to the next LTS I can hopefully test binary installs for arm architectures. Would be super cool if this works well off the shelf. So far UbuntuAsahi has honestly been awesome.

eslerm, 2024-02-12:
I wonder if schroots [0][1] would help if LXD does not? It's what I use to build binaries for different Ubuntu releases on my Mantic machine. Each shroot can use their own apt repos.

Please let me know if schoots and LXDs are a possibility.

[0] http://logan.tw/posts/2018/02/24/manage-chroot-environments-with-schroot/
[1] https://wiki.ubuntu.com/SecurityTeam/BuildEnvironment

peterdavidfagan, 2024-02-12:
Thanks @eslerm 🙏, I haven't used schroots before I'm going to check it out.

eslerm, 2024-02-16:
Please let us know if that worked for you @peterdavidfagan 

We're getting a few requests for ROS 2 support (https://github.com/UbuntuAsahi/ubuntu-asahi/issues/95) and I'd like to help find a solution :)

@cskn95 would you be able to test a jammy schroot on Ubuntu Asahi 23.10 for your use case?

cskn95, 2024-02-16:
Unfortunately I got errors on both ROS2 and Gazebo.

This is for ROS2:
`(jammy)ubuntu@localhost:~$ ros2 doctor
/opt/ros/iron/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: rcpputils has been updated to a new version. local: 2.6.2 < latest: 2.6.3
/opt/ros/iron/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: rqt_py_common has been updated to a new version. local: 1.3.3 < latest: 1.3.4
/opt/ros/iron/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: rqt_gui_py has been updated to a new version. local: 1.3.3 < latest: 1.3.4
/opt/ros/iron/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: rqt_gui_cpp has been updated to a new version. local: 1.3.3 < latest: 1.3.4
/opt/ros/iron/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: rqt_gui has been updated to a new version. local: 1.3.3 < latest: 1.3.4
2024-02-16 04:50:10.405 [RTPS_TRANSPORT_SHM Error] Failed to create segment 4fe208d79fe7c4a9: Permission denied -> Function compute_per_allocation_extra_size
2024-02-16 04:50:10.405 [RTPS_TRANSPORT_SHM Error] Failed to create Shared Memory Manager for domain fastrtps: Permission denied -> Function create
2024-02-16 04:50:10.405 [RTPS_PARTICIPANT Error] Unable to Register SHM Transport. SHM Transport is not supported in the current platform. -> Function RTPSParticipantImpl

All 5 checks passed
` 

And this is for Gazebo:
`(jammy)ubuntu@localhost:~$ gz sim
qt.qpa.xcb: could not connect to display 
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

Stack trace (most recent call last):
#31   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabdd3247, in ruby_run_node
#30   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabdcf983, in 
#29   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabf54a47, in rb_vm_exec
#28   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabf4f81b, in 
#27   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabf4ce8b, in 
#26   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabf4ae83, in 
#25   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabe9b7e7, in 
#24   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabdd0f73, in rb_protect
#23   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabf58fdf, in rb_yield
#22   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabf54a47, in rb_vm_exec
#21   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabf4f81b, in 
#20   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabf4ce8b, in 
#19   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabf4ae83, in 
#18   Object "/usr/lib/aarch64-linux-gnu/ruby/3.0.0/fiddle.so", at 0xffffa8245d93, in 
#17   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabf19ae3, in rb_nogvl
#16   Object "/usr/lib/aarch64-linux-gnu/ruby/3.0.0/fiddle.so", at 0xffffa82456cb, in 
#15   Object "/lib/aarch64-linux-gnu/libffi.so.8", at 0xffffa8223a93, in 
#14   Object "/lib/aarch64-linux-gnu/libffi.so.8", at 0xffffa8226e0f, in 
#13   Object "/usr/lib/aarch64-linux-gnu/libgz-sim8-gz.so.8.1.0", at 0xffffa771acab, in runGui
#12   Object "/lib/aarch64-linux-gnu/libgz-sim8-gui.so.8", at 0xffffa7559c3f, in gz::sim::v8::gui::runGui(int&, char**, char const*, char const*, int, char const*, char const*)
#11   Object "/lib/aarch64-linux-gnu/libgz-sim8-gui.so.8", at 0xffffa7557b27, in gz::sim::v8::gui::createGui(int&, char**, char const*, char const*, bool, char const*, int, char const*, char const*)
#10   Object "/lib/aarch64-linux-gnu/libgz-sim8-gui.so.8", at 0xffffa7555363, in gz::sim::v8::gui::launchQuickStart(int&, char**, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#9    Object "/lib/aarch64-linux-gnu/libgz-gui8.so.8", at 0xffffa680a5fb, in gz::gui::Application::Application(int&, char**, gz::gui::WindowType, char const*)
#8    Object "/lib/aarch64-linux-gnu/libQt5Widgets.so.5", at 0xffffa62886b3, in QApplicationPrivate::init()
#7    Object "/lib/aarch64-linux-gnu/libQt5Gui.so.5", at 0xffffa51acbe7, in QGuiApplicationPrivate::init()
#6    Object "/lib/aarch64-linux-gnu/libQt5Core.so.5", at 0xffffa4de3f47, in QCoreApplicationPrivate::init()
#5    Object "/lib/aarch64-linux-gnu/libQt5Gui.so.5", at 0xffffa51a98db, in QGuiApplicationPrivate::createEventDispatcher()
#4    Object "/lib/aarch64-linux-gnu/libQt5Gui.so.5", at 0xffffa51a942b, in QGuiApplicationPrivate::createPlatformIntegration()
#3    Object "/lib/aarch64-linux-gnu/libQt5Core.so.5", at 0xffffa4bd0f63, in QMessageLogger::fatal(char const*, ...) const
#2    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0xffffabba712f, in abort
#1    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0xffffabbba67b, in raise
#0    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0xffffabbff200, in 
Aborted (Signal sent by tkill() 51681 1000)
Escalating to SIGKILL on [Gazebo Sim Server]
`

peterdavidfagan, 2024-02-16:
> Please let us know if that worked for you @peterdavidfagan
> We're getting a few requests for ROS 2 support (https://github.com/UbuntuAsahi/ubuntu-asahi/issues/95) and I'd like to help find a solution :)

I have been able to successfully build ROS 2 packages from source on Ubuntu Asahi. This being said, it has been challenging as many binaries aren't easily installed. Also I have limited myself to packages that are compatible with Apple metal GPU.

eslerm, 2024-02-18:
I had to delete [@cskn95](https://github.com/cskn95)'s comment from two days ago, since the message contained strings which GitHub thought were references to other issues.

Below is the deleted comment with minor reformatting:

Unfortunately I got errors on both ROS2 and Gazebo.

This is for ROS2:
```
(jammy)ubuntu@localhost:~$ ros2 doctor
/opt/ros/iron/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: rcpputils has been updated to a new version. local: 2.6.2 < latest: 2.6.3
/opt/ros/iron/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: rqt_py_common has been updated to a new version. local: 1.3.3 < latest: 1.3.4
/opt/ros/iron/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: rqt_gui_py has been updated to a new version. local: 1.3.3 < latest: 1.3.4
/opt/ros/iron/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: rqt_gui_cpp has been updated to a new version. local: 1.3.3 < latest: 1.3.4
/opt/ros/iron/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: rqt_gui has been updated to a new version. local: 1.3.3 < latest: 1.3.4
2024-02-16 04:50:10.405 [RTPS_TRANSPORT_SHM Error] Failed to create segment 4fe208d79fe7c4a9: Permission denied -> Function compute_per_allocation_extra_size
2024-02-16 04:50:10.405 [RTPS_TRANSPORT_SHM Error] Failed to create Shared Memory Manager for domain fastrtps: Permission denied -> Function create
2024-02-16 04:50:10.405 [RTPS_PARTICIPANT Error] Unable to Register SHM Transport. SHM Transport is not supported in the current platform. -> Function RTPSParticipantImpl

All 5 checks passed
```

And this is for Gazebo:
```
(jammy)ubuntu@localhost:~$ gz sim
qt.qpa.xcb: could not connect to display 
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

Stack trace (most recent call last):
#31   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabdd3247, in ruby_run_node
#30   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabdcf983, in 
#29   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabf54a47, in rb_vm_exec
#28   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabf4f81b, in 
#27   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabf4ce8b, in 
#26   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabf4ae83, in 
#25   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabe9b7e7, in 
#24   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabdd0f73, in rb_protect
#23   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabf58fdf, in rb_yield
#22   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabf54a47, in rb_vm_exec
#21   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabf4f81b, in 
#20   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabf4ce8b, in 
#19   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabf4ae83, in 
#18   Object "/usr/lib/aarch64-linux-gnu/ruby/3.0.0/fiddle.so", at 0xffffa8245d93, in 
#17   Object "/lib/aarch64-linux-gnu/libruby-3.0.so.3.0", at 0xffffabf19ae3, in rb_nogvl
#16   Object "/usr/lib/aarch64-linux-gnu/ruby/3.0.0/fiddle.so", at 0xffffa82456cb, in 
#15   Object "/lib/aarch64-linux-gnu/libffi.so.8", at 0xffffa8223a93, in 
#14   Object "/lib/aarch64-linux-gnu/libffi.so.8", at 0xffffa8226e0f, in 
#13   Object "/usr/lib/aarch64-linux-gnu/libgz-sim8-gz.so.8.1.0", at 0xffffa771acab, in runGui
#12   Object "/lib/aarch64-linux-gnu/libgz-sim8-gui.so.8", at 0xffffa7559c3f, in gz::sim::v8::gui::runGui(int&, char**, char const*, char const*, int, char const*, char const*)
#11   Object "/lib/aarch64-linux-gnu/libgz-sim8-gui.so.8", at 0xffffa7557b27, in gz::sim::v8::gui::createGui(int&, char**, char const*, char const*, bool, char const*, int, char const*, char const*)
#10   Object "/lib/aarch64-linux-gnu/libgz-sim8-gui.so.8", at 0xffffa7555363, in gz::sim::v8::gui::launchQuickStart(int&, char**, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#9    Object "/lib/aarch64-linux-gnu/libgz-gui8.so.8", at 0xffffa680a5fb, in gz::gui::Application::Application(int&, char**, gz::gui::WindowType, char const*)
#8    Object "/lib/aarch64-linux-gnu/libQt5Widgets.so.5", at 0xffffa62886b3, in QApplicationPrivate::init()
#7    Object "/lib/aarch64-linux-gnu/libQt5Gui.so.5", at 0xffffa51acbe7, in QGuiApplicationPrivate::init()
#6    Object "/lib/aarch64-linux-gnu/libQt5Core.so.5", at 0xffffa4de3f47, in QCoreApplicationPrivate::init()
#5    Object "/lib/aarch64-linux-gnu/libQt5Gui.so.5", at 0xffffa51a98db, in QGuiApplicationPrivate::createEventDispatcher()
#4    Object "/lib/aarch64-linux-gnu/libQt5Gui.so.5", at 0xffffa51a942b, in QGuiApplicationPrivate::createPlatformIntegration()
#3    Object "/lib/aarch64-linux-gnu/libQt5Core.so.5", at 0xffffa4bd0f63, in QMessageLogger::fatal(char const*, ...) const
#2    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0xffffabba712f, in abort
#1    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0xffffabbba67b, in raise
#0    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0xffffabbff200, in 
Aborted (Signal sent by tkill() 51681 1000)
Escalating to SIGKILL on [Gazebo Sim Server]
```

eslerm, 2024-04-25:
22.04 is now part of the installer, but you need to set `EXPERT=1` for the option.

Ubuntu Asahi is not supporting 22.04 beyond this.

closed 2024-04-25 by eslerm

asimonov, 2024-11-22:
hello, I am trying to run ROS2 humble + Gazebo Harmonic on Asahi 24.04. 

I am doing this inside docker which is ubuntu22.04, but I install `sudo add-apt-repository ppa:tobhe/asahi`.

when trying to run Gazebo I see the following errors:

```
spaceros-user@m1asahi:~$ gz sim
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-spaceros-user'
UABI mismatch: Kernel 10011, Mesa 10009
glx: failed to create dri3 screen
failed to load driver: asahi
UABI mismatch: Kernel 10011, Mesa 10009
glx: failed to create dri3 screen
failed to load driver: asahi
[Err] [SystemPaths.cc:525] Could not resolve file [texture.png]
[Err] [SystemPaths.cc:525] Could not resolve file [texture.png]
[Err] [SystemPaths.cc:525] Could not resolve file [texture.png]
UABI mismatch: Kernel 10011, Mesa 10009
libEGL warning: egl: failed to create dri2 screen
UABI mismatch: Kernel 10011, Mesa 10009
libEGL warning: egl: failed to create dri2 screen
UABI mismatch: Kernel 10011, Mesa 10009
libEGL warning: egl: failed to create dri2 screen
[GUI] [Err] [SystemPaths.cc:525] Could not resolve file [texture.png]
[GUI] [Err] [SystemPaths.cc:525] Could not resolve file [texture.png]
[GUI] [Err] [SystemPaths.cc:525] Could not resolve file [texture.png]
```

and then it runs at 5fps, so OpenGL is not working.

but glxgears give me this inside same docker:

```
spaceros-user@m1asahi:~$ glxgears
UABI mismatch: Kernel 10011, Mesa 10009
glx: failed to create dri3 screen
failed to load driver: asahi
5610 frames in 5.0 seconds = 1121.871 FPS
```

outside of the docker i can run gazebo with OpenGL support at 60fps.

what may be the issue with docker setup?

asimonov, 2024-11-24:
> Please try adding https://launchpad.net/~ubuntu-asahi/+archive/ubuntu/mesa in your docker container, that has a newer mesa version that should work with the noble kernel.

that worked! 

on ubuntu22 inside docker I can run Gazebo Harmonic at 60fps. the host is Ubuntu Asahi 24.

question - can I possibly use this docker image in an Ubuntu24 VM on Macos (like Parallels or VM fusion)? 
would I need to update kernel to 6.11? 
or can I install Asahi in a VM by any chance?

tobhe, 2024-11-24:
You don't need Asahi for VMs, Asahi is all about native support. You can install a regular Ubuntu arm64 iso in your VM and run docker in that. The VM only interacts with a virtual GPU so the mesa part in the container would not be needed in such a setup.

asimonov, 2024-11-24:
> You don't need Asahi for VMs, Asahi is all about native support. You can install a regular Ubuntu arm64 iso in your VM and run docker in that. The VM only interacts with a virtual GPU so the mesa part in the container would not be needed in such a setup.

Yes, I know. But it would be cool to use Asahi's OpenGL inside VMs on Macos. will allow saving time not booting into different systems.
Do you think it is possible?

tobhe, 2024-11-24:
No, because the VM does have real access to the physical Apple GPU which the Asahi driver needs. In the VM you get a paravirtualized GPU using a different driver. It might still support OpenGL though.