UbuntuAsahi/ubuntu-asahi #91: Mac mini M2 free space blocked

ahmeier, 2024-02-11:
I tried to install Ubuntu using the install script, but even after removing all local timemachine snapshots and completing all software updates the free space on the internal HD remains blocked. I have a 1TB internal HD and there is 380GB of free space when checking on disk utility. I can add additional APFS containers, but the install script complains of overhead blocking the free space. It seems that for some reason it is impossible to access the existing free space.

Hardware:
Mac Mini M2 model 13,14

Any suggestions what else to do?

tobhe, 2024-02-11:
Hi! Hard to tell what exactly is wrong without some more information. Can you show us the output of `diskutil list`?

ahmeier, 2024-02-11:
Thanks for the quick response. I attached the diskutil printout for the internal disk. Looks like the APFS container includes the whole disk, including the free space, but don’t think the container can be shrunk. Perhaps that is the problem.Here is the printout:/dev/disk0 (internal, physical):   #:                       TYPE NAME                    SIZE       IDENTIFIER   0:      GUID_partition_scheme                        *1.0 TB     disk0   1:             Apple_APFS_ISC Container disk1         524.3 MB   disk0s1   2:                 Apple_APFS Container disk3         994.7 GB   disk0s2   3:        Apple_APFS_Recovery Container disk2         5.4 GB     disk0s3/dev/disk3 (synthesized):   #:                       TYPE NAME                    SIZE       IDENTIFIER   0:      APFS Container Scheme -                      +994.7 GB   disk3                                 Physical Store disk0s2   1:                APFS Volume Macintosh HD            10.1 GB    disk3s1   2:              APFS Snapshot com.apple.os.update-... 10.1 GB    disk3s1s1   3:                APFS Volume Preboot                 6.0 GB     disk3s2   4:                APFS Volume Recovery                909.4 MB   disk3s3   5:                APFS Volume Data                    632.8 GB   disk3s5   6:                APFS Volume VM                      1.1 GB     disk3s6Sent from my iPadAndreas H Meier MD Dr med MEd FACS FAAPMichael & Rissa Ratner Professor of Pediatric SurgeryGeneral Surgery Residency Program Director and Vice Chair for EducationDepartment of ***@***.***

eslerm, 2024-02-11:
@ahmeier the attached logs did not format well. Could you please fix-up the markdown in your comment?

Disabling Time Machine and [Asahi's Partition Cheatsheet](https://github.com/AsahiLinux/docs/wiki/Partitioning-cheatsheet) may help.

ahmeier, 2024-02-12:
Sorry, sent comment by email, here is the better formatted output:
/dev/disk0 (internal, physical):
  | #: |TYPE NAME                                             |SIZE.           | IDENTIFIER|
  | 0: |GUID_partition_scheme                         |*1.0 TB        |   disk0
  | 1: |Apple_APFS_ISC Container disk1          | 524.3 MB   |    disk0s1
  | 2:|Apple_APFS Container disk3                  | 994.7 GB   |   disk0s2
  | 3:|Apple_APFS_Recovery Container disk2| 5.4 GB          disk0s3

/dev/disk3 (synthesized):
   #:  TYPE NAME                                                   SIZE              IDENTIFIER
   0:  APFS Container Scheme -                            +994.7 GB   disk3
                                 Physical Store disk0s2
   1: APFS Volume Macintosh HD                           10.1 GB        disk3s1
   2: APFS Snapshot com.apple.os.update-...      10.1 GB        disk3s1s1
   3: APFS Volume Preboot                                      6.0 GB        disk3s2
   4: APFS Volume Recovery                                   909.4 MB   disk3s3
   5: APFS Volume Data                                           632.8 GB    disk3s5
   6: APFS Volume VM                                                  1.1 GB     disk3s6

tobhe, 2024-02-14:
I can't spot anything obviously wrong in there. As @eslerm said it might have to do with time machine, see https://github.com/AsahiLinux/asahi-installer/issues/86