UbuntuAsahi/ubuntu-asahi #92: ubuntu 23.10 night light

dooraim, 2024-02-11:
I'm using the following version of Ubuntu and Asahi Linux Kernel,

```bash
ubuntu@macbookm1pro:~$ cat /etc/os-release 
PRETTY_NAME="Ubuntu 23.10"
NAME="Ubuntu"
VERSION_ID="23.10"
VERSION="23.10 (Mantic Minotaur)"
VERSION_CODENAME=mantic
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=mantic
LOGO=ubuntu-logo
ubuntu@macbookm1pro:~$ uname -a
Linux macbookm1pro 6.6.0-1003-apple-arm #3-Ubuntu SMP PREEMPT_DYNAMIC Wed Dec 27 01:40:02 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux
ubuntu@macbookm1pro:~$ 
```

I can not using Night Light, also if I have GPU driver installed. 

There is any solution about this issue? Or is there a 3rd parts software to install that make the same functions?

eslerm, 2024-02-14:
This is likely a duplicate of https://github.com/UbuntuAsahi/ubuntu-asahi/issues/53

For these issues I'll do some testing over the weekend.

[wlsunset](https://sr.ht/~kennylevinsen/wlsunset/) might be a solution.

eslerm, 2024-02-18:
Thanks for testing @dooraim :pray: 

I posted some followup in https://github.com/UbuntuAsahi/ubuntu-asahi/issues/53#issuecomment-1950992211

I'm going to close this issue as a duplicate to keep the conversation in the original thread.

closed 2024-02-18 by eslerm