UbuntuAsahi/ubuntu-asahi #93: Restart loop

0Yuuw, 2024-02-13:
Hello,

I have a MacBook Air M2, and since I installed Ubuntu Asahi, it doesn't start and just loops the booting process. :-( When I tried to access the boot menu, it says to go to support.apple.com/mac/restore, but it requires a second Mac, which I don't have. Does anyone know how I can solve this, or do I need to go to the Apple Store?

tobhe, 2024-02-13:
Can you explain in more detail what exactly you did before? What happens if you hold down power during boot? The installer asks you to resize your APFS partition, what size did you choose?

0Yuuw, 2024-02-13:
I allocated 30% of the disk space for macOS and the remainder for Ubuntu. When I try to access the boot menu by holding the power button, it displays an error message directing me to the Apple website. Otherwise, it just keeps restarting over and over again

0Yuuw, 2024-02-17:
Problem solved, I borrowed my friend's mac and reset my mac!

closed 2024-02-17 by 0Yuuw