UbuntuAsahi/ubuntu-asahi #94: your system did not boot into correct recovery

cskn95, 2024-02-14:
I'm using M2 Air with Sonoma 14.3.1 and tried to install Ubuntu 23.10. When the machine has shut down and I reopen after a minute and choose Ubuntu on startup menu, a login page appears and after that in a terminal it says "your system did not boot into correct recovery" and wants my login info to shutdown. I tried 3 times but no changes.

tobhe, 2024-02-15:
Hi! Please try booting into macos and run the installer a second time. It should give you a repair option.
Let us know if the problem persists after the repair!

cskn95, 2024-02-15:
> Hi! Please try booting into macos and run the installer a second time. It should give you a repair option. Let us know if the problem persists after the repair!

I tried but nothing changed. I tried repair option 2 times and even tried to install Fedora Remix version from Asahi Linux web page (before I completely removed Ubuntu Asahi) but nothing changes. I'm doing exactly what it says, I even watched videos and nope, I'm doing it right, sure about it.

tobhe, 2024-02-15:
Weird. Does a fresh Fedora install work? This is new to me, I'll try and run a fresh install on my test hardware.

cskn95, 2024-02-15:
3rd installation attempt worked fine interestingly. Sorry to bother you, everything seems like works fine.

tobhe, 2024-02-15:
Thank you for the report! We rely on our community to notify us if anything goes wrong because we can't test every possible combination of hardware and software ourselves. Your feedback is very much appreciated :) Glad it turned out not to be a bigger issue.

closed 2024-02-15 by tobhe