UbuntuAsahi/ubuntu-asahi #95: Using GPU in Distrobox

cskn95, 2024-02-15:
Hello again! I want to use ROS2/Gazebo but they both need Ubuntu 22.04 for installation. I tried to trick install command and passed version check but installation threw me errors with python packages, python_vcstools for ROS2 Iron and Python 3.11 for Gazebo. 

I tried using Distrobox and installed Ubuntu 22.04 and installed ROS2 Iron and Gazebo Harmonic fine but this time Gazebo threw me GPU driver issues.

How can I use 22.04 apps with GPU acceleration? Gazebo needs OpenGL 3.3 at least.

tobhe, 2024-02-16:
try adding https://launchpad.net/~tobhe/+archive/ubuntu/asahi via `sudo add-apt-repository ppa:tobhe/asahi && sudo apt update && suo apt upgrade` and see if that helps. Maybe distrobox also needs to be configured to actually use the GPU but I don't know that, never used it.

cskn95, 2024-02-16:
Driver issues now gone thanks for that but still when I try to open a project or create one it says "Gazebo GUI is not responding" and gives me this errors:
`[GUI] [Wrn] [Application.cc:904] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[Wrn] [Component.hh:144] Trying to serialize component with data type [N3sdf3v145WorldE], which doesn't have `operator<<`. Component will not be serialized.

`