UbuntuAsahi/ubuntu-asahi #97: (Please help) Reclaim free space for MacOS

MWRevamped, 2024-03-02:
I'm not that familiar with operating systems for computers other than Windows and OSX, but I've always wanted to try some out, and I set Ubuntu up without failure but when I freed up some storage and repartitioned my MacOS volume for it, instead of merging the free space, it exists as a separate volume, and if I can't merge the space I'd like to reclaim it

[image]
[image]
[image]
[image]

eslerm, 2024-03-02:
This operation is not simple to perform. First you would need to resize the apfs containers, then the asahi partition, and then the asahi filesystem, in that order. This would require another linux os for some of the steps, likely with a live cd in this case. I can't vouch for [this guide](https://www.reddit.com/r/AsahiLinux/comments/10xn538/guide_how_to_resize_your_asahi_root_partition/) but it seems to be on the right track.

I recommend backing up, wiping, and reinstalling.

closed 2024-03-02 by eslerm

MWRevamped, 2024-03-02:
Aw alright, thank you for the response tho, have a good day!