UbuntuAsahi/ubuntu-asahi #98: M2 Pro kernel 6.6.0

1Guardian, 2024-03-02:
Hello, 

I updated my ubuntu version to 23.10 and along with it the kernel updated from 6.5.0 to 6.6.0. However, when booted with the new kernel, only software rendering is working, and checking the dmesg log notes: Incompatible devicetree as well as audio route errors from the ASoC module.

Does the M2 pro need a different kernel or did the update possible get botched somehow?

eslerm, 2024-03-02:
6.6.0 is not yet support by Ubuntu Asahi.

If you are using experimental 6.6.0 https://launchpad.net/~john-cabaj/+archive/ubuntu/apple-arm you need an updated mesa https://launchpad.net/~eslerm/+archive/ubuntu/asahi-testing

tobhe, 2024-03-02:
I don't think that is quite right, we do ship 6.6.0-1003.3  for mantic.

@1Guardian can you share us what `$ apt list --installed | grep mesa` and `$ apt list --installed | grep apple-arm` shows on your machine?

eslerm, 2024-03-03:
@1Guardian could you please share `uname -a` as well?

6.6.0-1003 ran smoothly for me, but 6.6.0-1004 requires new mesa drivers.

@tobhe I don't believe we official support 6.6.0 yet. iiuc, it exists in a [test ppa](https://launchpad.net/~ubuntu-asahi/+archive/ubuntu/ubuntu-asahi?field.series_filter=mantic) and not the [ubuntu asahi ppa](https://launchpad.net/~ubuntu-asahi/+archive/ubuntu/ubuntu-asahi?field.series_filter=mantic).

tobhe, 2024-03-03:
@eslerm but the PPA we currently use until we found a suitable migration strategy is still https://launchpad.net/~tobhe/+archive/ubuntu/asahi

Migrating to the team ppa is a future goal.

1Guardian, 2024-03-04:
Sorry, I ended up just reinstalling yesterday to fix it since it was just a test env. After reinstall all was fine so I'm assuming there was some sort of issue in the upgrade rather than the kernel & packages themselves.

tobhe, 2024-04-29:
Closing because this seems to be resolved for now. Thanks!

closed 2024-04-29 by tobhe