UbuntuAsahi/ubuntu-asahi #99: OpenGL 4.6 support?

aaaaaammmmmm, 2024-03-03:
Hi, any plans to include OpenGL 4.6 support like Fedora Asahi got a few weeks back?
Maybe there's already a way to install it?
Thanks!
[https://rosenzweig.io/blog/conformant-gl46-on-the-m1.html](https://rosenzweig.io/blog/conformant-gl46-on-the-m1.html)

eslerm, 2024-03-04:
@tobhe could you please copy @john-cabaj's [6.6.0-1004 kernel packages](https://launchpad.net/~john-cabaj/+archive/ubuntu/apple-arm) and [libdrm and mesa from my ppa](https://launchpad.net/~eslerm/+archive/ubuntu/asahi-testing)? All of these packages need to be upgraded together.

```
eslerm@moon:~$ uname -a
Linux moon 6.6.0-1004-apple-arm #4-Ubuntu SMP PREEMPT_DYNAMIC Fri Feb 16 04:23:28 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux
eslerm@moon:~$ glxinfo|grep 'OpenGL ver'
OpenGL version string: 4.6 (Compatibility Profile) Mesa 24.0~pre20240228-0asahi3.23.10
```
[image]

tobhe, 2024-03-04:
@eslerm I will publish them to asahi-testing but we still need 22.04 mesa binaries before we can officially publish them.

eslerm, 2024-03-05:
I packaged a new mesa driver for jammy. If that's all that's needed for gnome, everything should be ready.

It does not look possible to install linux-apple-arm 6.6.0-1004 on jammy without bumping libc 💀

```
ubuntu@localhost:~$ apt-cache policy linux-apple-arm
linux-apple-arm:
  Installed: 6.5.0.1009.11~22.04.5
  Candidate: 6.6.0.1004.4
  Version table:
     6.6.0.1004.4 500
        500 https://ppa.launchpadcontent.net/john-cabaj/apple-arm/ubuntu jammy/main arm64 Packages
     6.5.0.1013.15~22.04.9 500
        500 http://ppa.launchpad.net/tobhe/asahi/ubuntu jammy/main arm64 Packages
 *** 6.5.0.1009.11~22.04.5 100
        100 /var/lib/dpkg/status
ubuntu@localhost:~$ sudo apt upgrade linux-apple-arm
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 linux-headers-6.6.0-1004-apple-arm : Depends: linux-apple-arm-headers-6.6.0-1004 but it is not going to be installed
                                      Depends: libc6 (>= 2.38) but 2.35-0ubuntu3.6 is to be installed
ubuntu@localhost:~$ uname -a
Linux localhost.localdomain 6.5.0-1009-apple-arm #11~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Oct 12 06:30:14 UTC 2 aarch64 aarch64 aarch64 GNU/Linux
```

tobhe, 2024-03-05:
good to know. we won't need the kernel on jammy for now so that's not a problem

eslerm, 2024-03-05:
ROS users will need to apt pin the previous mesa version. I couldn't add 6.6.0-1004 as a dependency to jammy's mesa ofc.

tobhe, 2024-03-05:
@eslerm we don't need to publish it to the regular ppa for 22.04. We just need it to go into the snap.

tobhe, 2024-04-29:
This should be fixed now. Thank you!

closed 2024-04-29 by tobhe