UbuntuAsahi/ubuntu-asahi #100: Display output is not active

OumarBalde04, 2024-03-14:
These are my issues I don't know where is my fault

[image]
[image]

OumarBalde04, 2024-03-14:
for the record I got a MacBook Air m2 2023

tobhe, 2024-03-14:
Looks like you are trying to boot something in QEMU. Ubuntu Asahi provides bare metal Ubuntu images for Apple Silicon macs, we can't help you with running Ubuntu in a VM. Regular Ubuntu ARM installer images are available at https://ubuntu.com/download/server/arm