UbuntuAsahi/ubuntu-asahi #101: Brightness Monitor Control

maitreverge, 2024-03-26:
Hi ! 

I've been searching around for quite some time different solutions to control my external monitor brightness.

I tried :
updating `grub` settings
installing `kubuntu desktop`
`xrandr` command,
`brightness-controller` package
`light` package
`brightnessctl` package
`brightness-controller` package

I am running out of ideas, and my monitor refuse to cooperate regarding self brightness control.

I tweaked every above solutions, read package documentation, and failed to reach the brightness control.

Hardware : Mac Mini M1.
Ubuntu 23.01

Hope someone could help !

eslerm, 2024-03-28:
`ddcutil` might work

I don't have a Mini to test

maitreverge, 2024-03-28:
Did not work, but thanks for showing me a precious package ! I guess I'll wait for a native support....

eslerm, 2024-05-19:
@maitreverge do you have any success changing the brightness with `echo 32 | sudo tee /sys/class/backlight/X/brightness ` where `32` is a number between 0 and 256 and `X` is the external display?

maitreverge, 2024-05-21:
@eslerm Thanks for your interest in the issue !
I did try your take, and the directory `/sys/class/backlight/` is empty.

I honestly gave up, and told myself that not having backlight control is a tradeoff for having Ubuntu on a Mac M1, _or such feature is maybe not available yet..._

I appreciate your answer tho, I'll close the issue from now, hopping for something to happen !

Thanks for your time !

closed 2024-05-21 by maitreverge