UbuntuAsahi/ubuntu-asahi #102: I CAN not install this no matter what!!

astroskyoffical, 2024-04-04:
[log.txt](https://github.com/UbuntuAsahi/ubuntu-asahi/files/14860512/log.txt)

log attached, i have enough freespace and no matter what i do i cant proceed to install this ! its driving me nuts and id appreciate a more keen eye to help me out.  please someone?

eslerm, 2024-04-04:
@tobhe [mentioned that the file server might be down today](https://oftc.irclog.whitequark.org/ubuntu-asahi/2024-04-03) and your log.txt suggest that downloading the image file failed.

Hopefully it will just work after tobhe gives the all clear tomorrow :)

tobhe, 2024-04-04:
Indeed, looks like something went wrong when moving some things to a new host. Should be fixed now, thanks for letting us know @astroskyoffical !

astroskyoffical, 2024-04-04:
unfortunately even after reinstalling......its the same issue....      

Last login: Thu Apr  4 05:25:25 on console
astro@Austins-MacBook-Air ~ % curl -sL https://ubuntuasahi.org/install | sh
Bootstrapping installer:
  Checking version...
  Version: v0.7.1
  Downloading...
  Extracting...
  Initializing...

The installer needs to run as root.
Please enter your sudo password if prompted.
Password:

Welcome to the Asahi Linux installer!

This installer will guide you through the process of setting up
Asahi Linux on your Mac.

Please make sure you are familiar with our documentation at:
  https://alx.sh/w

Press enter to continue.

Collecting system information...
  Product name: MacBook Air (M1, 2020)
  SoC: Apple M1
  Device class: j313ap
  Product type: MacBookAir10,1
  Board ID: 0x26
  Chip ID: 0x8103
  System firmware: iBoot-10151.41.12
  Boot UUID: 7DAD646C-13CF-4316-A62B-9A5EB7667215
  Boot VGID: 7DAD646C-13CF-4316-A62B-9A5EB7667215
  Default boot VGID: 7DAD646C-13CF-4316-A62B-9A5EB7667215
  Boot mode: macOS
  OS version: 14.1 (23B74)
  OS restore version: 23.2.74.0.0,0
  Main firmware version: 14.1 (23B74)
  No Fallback System Firmware / rOS
  SFR version: 23.2.74.0.0,0
  SystemRecovery version: 23.2.74.0.0,0 (14.1 23B74)
  Login user: astro

Collecting partition information...
  System disk: disk0

Collecting OS information...

Partitions in system disk (disk0):
  1: APFS [Macintosh HD] (245.11 GB, 6 volumes)
    OS: [B*] [Macintosh HD] macOS v14.1 [disk3s1s1, 7DAD646C-13CF-4316-A62B-9A5EB7667215]
  2: APFS (System Recovery) (5.37 GB, 2 volumes)
    OS: [  ] recoveryOS v14.1 [Primary recoveryOS]

  [B ] = Booted OS, [R ] = Booted recovery, [? ] = Unknown
  [ *] = Default boot volume

Using OS 'Macintosh HD' (disk3s1s1) for machine authentication.

Choose what to do:
  r: Resize an existing partition to make space for a new OS
  q: Quit without doing anything
» Action (r): r

We're going to resize this partition:
  APFS [Macintosh HD] (245.11 GB, 6 volumes)
  Total size: 245.11 GB
  Free space: 225.64 GB
[... 222 lines cut ...]
Choose an OS to install:
  1: Ubuntu Desktop 23.10
» OS: 1

Downloading OS package info...
root        : ERROR    Exception caught
Traceback (most recent call last):
  File "/private/tmp/asahi-install/main.py", line 1058, in <module>
    InstallerMain(installer_version).main()
  File "/private/tmp/asahi-install/main.py", line 866, in main
    while self.main_loop():
  File "/private/tmp/asahi-install/main.py", line 1021, in main_loop
    return self.action_install_into_free(parts_free)
  File "/private/tmp/asahi-install/main.py", line 292, in action_install_into_free
    self.osins.load_package()
  File "/private/tmp/asahi-install/osinstall.py", line 47, in load_package
    self.ucache = urlcache.URLCache(package)
  File "/private/tmp/asahi-install/urlcache.py", line 26, in __init__
    self.size = self.get_size()
  File "/private/tmp/asahi-install/urlcache.py", line 75, in get_size
    return int(res.getheader("Content-length"))
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'
If you need to file a bug report, please attach the log file:
  /private/tmp/asahi-install/installer.log
astro@Austins-MacBook-Air ~ %

tobhe, 2024-04-04:
Probably just your DNS cache being outdated, you might have to flush it or wait a bit longer. I verified it works here.

eslerm, 2024-04-17:
@astroskyoffical any luck?

tobhe, 2024-04-29:
Closing this one since we believe the issue is fixed. Thanks!

closed 2024-04-29 by tobhe