UbuntuAsahi/ubuntu-asahi #103: Few begginer issues

7sparrow, 2024-04-14:
I just installed the latest build and it went really well! Everything run's smoothly except for a few things post system setup.

My software updated is crashing everytime, even on restart. Im not sure how to troubleshoot it. Any tips/docs?

The keyboard backlight isnt working. Im on an M1 - should it work automatically? What should i check/do otherwise?

GFX drivers - Seem's we cant enable night light due to the current driver. Is there a workaround?

Thanks and great effort so far!

eslerm, 2024-04-16:
Hello @7sparrow o/

For software updates, could you try this in a terminal `sudo apt update && sudo apt upgrade` ?

Keyboard backlight came up recently in the irc. From kaazoo advised: "Gnome has a keyboard brightness control in the menu on the top right of the primary screen, where you can also change wifi, sound volume and so on. " or try enabling it with `brightnessctl` from a terminal.

For night light, most DEs have issues on arm64 except KDE. More info on: https://github.com/UbuntuAsahi/ubuntu-asahi/issues/53

7sparrow, 2024-04-16:
Yap, works perfectly fine now. Upgrade/update works but the software updater is still crashing.

I dont know how i missed the keyboard section. I cant tell if it appeared after i installed brightnessctl if not. Either way, its also working.

jitunayak, 2024-04-27:
> Yap, works perfectly fine now. Upgrade/update works but the software updater is still crashing.
> 
> I dont know how i missed the keyboard section. I cant tell if it appeared after i installed brightnessctl if not. Either way, its also working.

Is there a way to have timeout for backlight just like mac ? currently if you turn on, it just stays on infinitely

tobhe, 2024-04-29:
Thanks, I managed to reproduce the update-manager crash issue on a fresh 23.10 installation. I'll see if I can find out what's wrong.

tobhe, 2024-04-30:
For a temporary fix you can `sudo apt reinstall dbus`. I am still investigating why this is needed. Probably something broken in the image we currently ship.

tobhe, 2024-05-04:
Should also be fixed in the newest installer images now.