UbuntuAsahi/ubuntu-asahi #105: Can't log in after reboot

Pathsis, 2024-04-16:
Greetings! 

I installed Ubuntu Asahi on my M1 Macbook Pro. After logging in to the system for the first time, I only installed a language support, then I rebooted the system, and then, I could not log in to the system again. On the login screen, there is no session box available to enter the login password.

[image]

I tried pressing `fn` + `control` + `option` + `f3`, but all I get is a black page with `_` blinking and no response to typing any characters.

Also, when booting up, when the big Ubuntu icon appears in the center of the screen, there's a line in the upper right corner that says `failed to load "/"` (I don't remember it very well.)

What should I do? Do I need to reinstall?

Pathsis, 2024-04-16:
I've seen a few people from Reddit's asahi channel have this problem as well. The problem seems to be specific to Gnome. Following the information I got there, I was able to see the username appear after waiting for ten minutes, but still couldn't see the input box to enter a password.

Indeed, I find that Ubuntu Asahi seems to be an experimental program at the moment. I can't use it at all.

Pathsis, 2024-04-17:
> Could you please link the Reddit url? > > Does this affect upstream Asahi Linux? If so, is there a bug filled with Gnome as well?

Thank you for your reply! Here are the links I found for similar issues: 

https://discussion.fedoraproject.org/t/macbook-m1-pro-2021-gnome-login-freezes/98018 

https://www.reddit.com/r/AsahiLinux/comments/1815clc/unresponsive_login_screen_in_fedora_asahi/

 One of the links mentions that the problem may be related to keyboard dimming. Indeed, I remember that I did adjust the keyboard lights after I first installed and entered the system.

Pathsis, 2024-05-08:
I'm glad to see that my problem is getting some attention. I've heard from other sources that the problem may be related to the keyboard light driver, as I was adjusting the keyboard light at the time and the system understood that it froze.And the problem remains Gnome specific,

tobhe, 2024-05-08:
It looks like the keyboard backlight is fixed in a gnome stable branch. We might be able to backport the fix to mantic.

Meanwhile we do support release upgrades to 24.04 now which comes with gnome 46 and it looks like that might not be affected.

Thanks for digging up the additional links, that helped immensely!