UbuntuAsahi/ubuntu-asahi #106: zipfile.BadZipFile: File is not a zip file

Mravinko, 2024-04-17:
MacBook Pro 2021, installation stopped right after I chose OS type with `zipfile.BadZipFile: File is not a zip file` error.

[installer.log](https://github.com/UbuntuAsahi/ubuntu-asahi/files/15015881/installer.log)

eslerm, 2024-04-17:
From your logs, it looks like you are using the old installer link (`INSTALLER_DATA`).

Could you please try `curl -sL https://ubuntuasahi.org/install | sh` ?

If that does not work, please report *and* try again in a few hours. There was a server change and uptime has been lower than usual.

PhilStollery, 2024-04-17:
[installer.log](https://github.com/UbuntuAsahi/ubuntu-asahi/files/15016245/installer.log)

eslerm, 2024-04-17:
Thanks @PhilStollery 

@tobhe is out sick and likely won't be able to work on this for a little while

PhilStollery, 2024-04-17:
> Thanks @PhilStollery
> 
> @tobhe is out sick and likely won't be able to work on this for a little while

He must be keeping an eye on this - it's now fixed.

Mravinko, 2024-04-17:
Thanks for quick fix

closed 2024-04-17 by Mravinko

tobhe, 2024-04-18:
Thanks for the reports everyone, managed to deploy a quick fix yesterday. I'll have to come up with a better solution once I find more time.