UbuntuAsahi/ubuntu-asahi #107: stucked on M2 Ultra

zhx999, 2024-04-18:
When the installation finished and I tried to start the Ubuntu, the initialization process freezed

tobhe, 2024-04-18:
we will need more information to help you. can you maybe provide a screenshot? What did you do exactly and at which point did it get stuck? Did you follow the instructions and hold the power button to boot into the boot picker and then selected Ubuntu?

zhx999, 2024-04-18:
[image]

##### During running the install.sh, when resize the disk, here are some warnings:
------------------------------------------
Started APFS operation
Aligning shrink delta to 1,998,163,951,616 bytes and targeting a new container size of 1,998,165,377,024 bytes
Determined the minimum size for the APFS Container to be 1,193,078,161,408 bytes
Resizing APFS Container designated by APFS Container Reference disk3
The specific APFS Physical Store being resized is disk0s2
Verifying storage system
Using live mode
Performing fsck_apfs -n -x -l /dev/disk0s2
Checking the container superblock
Checking the checkpoint with transaction ID 4901858
Checking the space manager
Checking the space manager free queue trees
Checking the object map
Checking the encryption key structures
Checking volume /dev/rdisk3s1
Checking the APFS volume superblock
The volume Macintosh HD was formatted by newfs_apfs (1677.41.3.101.1) and last modified by apfs_kext (2236.101.1)
Checking the object map
Checking the snapshot metadata tree
Checking the snapshot metadata
Checking snapshot 1 of 1 (com.apple.os.update-39AFBADD5AD7CDAB000800931F501492F46ACCAF14B9622A5EFF21BDA87326B8, transaction ID 4219872)
Checking the fsroot tree
Checking the file extent tree
Checking the extent ref tree
Verifying volume object map space
The volume /dev/rdisk3s1 with UUID C856BF42-4F03-4AB1-8B1E-37EABEE0D144 appears to be OK
Checking volume /dev/rdisk3s2
Checking the APFS volume superblock
The volume Preboot was formatted by newfs_apfs (2142.120.7) and last modified by apfs_kext (2236.101.1)
Checking the object map
Checking the snapshot metadata tree
Checking the snapshot metadata
Checking the fsroot tree
Checking the extent ref tree
Verifying volume object map space
The volume /dev/rdisk3s2 with UUID D0D644E0-725D-4597-89A4-6A33DFAD726B appears to be OK
Checking volume /dev/rdisk3s3
Checking the APFS volume superblock
The volume Recovery was formatted by newfs_apfs (2142.120.7) and last modified by apfs_kext (2236.101.1)
Checking the object map
Checking the snapshot metadata tree
Checking the snapshot metadata
Checking the fsroot tree
Checking the extent ref tree
Verifying volume object map space
The volume /dev/rdisk3s3 with UUID D02DA538-DA07-4BC6-A69C-B6F7D3CC6240 appears to be OK
Checking volume /dev/rdisk3s4
Checking the APFS volume superblock
The volume Update was formatted by newfs_apfs (2142.120.7) and last modified by apfs_kext (2236.101.1)
Checking the object map
Checking the snapshot metadata tree
Checking the snapshot metadata
Checking the fsroot tree
Checking the extent ref tree
Verifying volume object map space
The volume /dev/rdisk3s4 with UUID 56ADE1A9-2095-4ED3-85C0-3BC67F77C63C appears to be OK
Checking volume /dev/rdisk3s5
Checking the APFS volume superblock
The volume Data was formatted by newfs_apfs (2142.120.7) and last modified by apfs_kext (2236.101.1)
Checking the object map
Checking the snapshot metadata tree
Checking the snapshot metadata
Checking the document ID tree
Checking the fsroot tree
warning: inode (id 1393491): Resource Fork xattr is missing for compressed file
warning: inode (id 1709556): Resource Fork xattr is missing for compressed file
warning: inode (id 19765154): Resource Fork xattr is missing for compressed file
warning: inode (id 19773055): Resource Fork xattr is missing for compressed file
warning: inode (id 26623288): Resource Fork xattr is missing for compressed file
warning: inode (id 27752216): Resource Fork xattr is missing for compressed file
warning: inode (id 30303258): Resource Fork xattr is missing for compressed file
warning: inode (id 34648062): Resource Fork xattr is missing for compressed file
Checking the extent ref tree
Checking the file key rolling tree
Verifying volume object map space
The volume /dev/rdisk3s5 with UUID 4B7D6277-3B16-4DA2-833C-30775DA79917 was found to be corrupt and needs to be repaired
Checking volume /dev/rdisk3s6
Checking the APFS volume superblock
The volume VM was formatted by apfs_boot_util (2142.120.7) and last modified by apfs_kext (2236.101.1)
Checking the object map
Checking the snapshot metadata tree
Checking the snapshot metadata
Checking the fsroot tree
Checking the extent ref tree
Verifying volume object map space
The volume /dev/rdisk3s6 with UUID 6DF6C9B7-BC6E-4225-8CEB-F2CC2C5C9C1A appears to be OK
Verifying allocated space
Performing deferred repairs
warning: need to clear bsd flags (0x20) in inode (object-id 1393491)
Skipped 8/8 repairs of this type in total
warning: found orphan/invalid xattr (id 1393491, name com.apple.decmpfs)
Skipped 8/8 repairs of this type in total
The container /dev/disk0s2 appears to be OK
Storage system check exit code is 0
Shrinking APFS Physical Store disk0s2 from 3,996,329,328,640 to 1,998,165,377,024 bytes
Shrinking APFS data structures
Shrinking partition
Modifying partition map
Finished APFS operation

Resize complete. Press enter to continue.

Collecting partition information...
  System disk: disk0

Collecting OS information...

---------------------------------------------

Other parts are fine.

Then I try to start the new Ubuntu system, it stucked at the initialization page, please see the attached screenshot:
[image]

I waited up to 1 min then held the power botton to the reboot page.

tobhe, 2024-04-19:
so the last screenshot clearly is systemd output, so your Linux kernel does seem to boot but then gets stuck.
Which hardware is this?
The warnings shown in the first screenshot do look like they might be the cause of your problems. `The volume /dev/rdisk3s5 with UUID 4B7D6277-3B16-4DA2-833C-30775DA79917 was found to be corrupt and needs to be repaired` in particular doesn't sound great.

t-komatu, 2024-04-19:
Sorry for intruding
I'm also having the same problem
It stops in the same place as the screen shot and remains the same even if I leave it for more than 10 minutes.
By the way, fedora 39 works without any problems.
The other thing I've tried is debian and it also stops in the same way.
I feel like there's a problem there

zhx999, 2024-04-19:
> so the last screenshot clearly is systemd output, so your Linux kernel does seem to boot but then gets stuck.
>
> Which hardware is this?
>
> The warnings shown in the first screenshot do look like they [...]

The hardware is M2 Ultra with 128GB RAM.

But if you check the rest of the information, it automatically detected and selected another disk without error. So I guess the volume allocation is ok, then the problem may be from the version of Linux system. E.g., I could try 22.04 not 23.10 the latest one.

The volume /dev/rdisk3s6 with UUID 6DF6C9B7-BC6E-4225-8CEB-F2CC2C5C9C1A appears to be OK
Verifying allocated space
Performing deferred repairs
warning: need to clear bsd flags (0x20) in inode (object-id 1393491)
Skipped 8/8 repairs of this type in total
warning: found orphan/invalid xattr (id 1393491, name com.apple.decmpfs)
Skipped 8/8 repairs of this type in total
The container /dev/disk0s2 appears to be OK
Storage system check exit code is 0

zhx999, 2024-04-19:
> Sorry for intruding
>
> I'm also having the same problem
>
> It stops in the same place as the screen shot and remains the same even if I leave it for more than 10 minutes.
>
> By the way, fedora 39 works [...]

Thanks for your comment, I guess you are right.

tobhe, 2024-04-29:
Could this be related to https://github.com/UbuntuAsahi/ubuntu-asahi/issues/108? Do you have HDMI connected while booting?

t-komatu, 2024-04-29:
I'm not the thread owner
I usually connect it via HDMI and start it up.
After seeing post #108, I reinstalled ubuntuasahi again on April 27th.
I tried removing the USB and HDMI, booting up, and then installing the HDMI after 10 minutes.
No video signal will be output from HDMI.
By the way, 23.10 is running on MacminiM2pro without any problems.

t-komatu, 2024-04-29:
sorry
Is it possible to install 24.04?
If there is a way to do it
I'll try it as soon as I get back

tobhe, 2024-04-29:
We added support for upgrading to 24.04 earlier today. For detailed instructions see https://github.com/UbuntuAsahi/ubuntu-asahi/issues/112#issuecomment-2082281936

t-komatu, 2024-04-30:
sorry
In my case it hangs during startup and does not accept any input.
Also, even if you start in safe mode
It may be a bit impossible as it does not accept apt update apt upgrade.
Well, I'll try it once I get back.
thank you very much.

tobhe, 2024-04-30:
ah you might need to wait a few days then. I am working on getting 24.04 installer images ready

t-komatu, 2024-05-01:
thank you
problem solved
apt update apt upgraed is working
Upgraded to 24.04 in safe mode
For the time being, I was able to log in on 24.04.
If a time is set for "screenblank" in "powersaving", when that time comes, the screen will go black and no keyboard input will be accepted.
Actually, there's no problem if you quickly change it to "Never".
Thank you for teaching me so much