UbuntuAsahi/ubuntu-asahi #108: Boot process hangs when HDMI cable is connected

HarmenJorritsma, 2024-04-21:
Hi! 

When HDMI cable is connected during boot process, the boot process hangs and throws a apple dcp (something with turn on...) message on a black screen. Whithout HDMI cable the boot process succeeds.

It is very remarkable that when HDMI cable is connected after boot process (when OS is running), HDMI connection works fine!

Any suggestions to solve this issue? It is very annoying to unplug and plug the HDMI cable every start up process.

Kind regards,

Harmen

tobhe, 2024-04-22:
thanks for the info, that is very good to know in case people report similar issues in the future! The next upstream Asahi kernel update apparently comes with a bunch of HDMI improvements, there is a fair chance it also fixes this.

 I'll let you know when we have a version to test but it might take a while.

devnoname120, 2024-04-25:
Interestingly I don't have this issue at all on Ubuntu Server 22.04 LTS.

tobhe, 2024-04-29:
I can confirm that there is not issue on my M1 mini 22.04, 23.10 and 24.04 so this seems to only affect laptops with HDMI ports. 24.04 ships with a 6.8 kernel, it would be worth checking if that fixes your issue.

wez, 2024-06-01:
To clarify what I see, sometimes I see just this:

[image]

sometimes I see the ubuntu logo and spinner wheel, but stopped.

I tried leaving hdmi disconnected for a while before connecting it, but I'm not able to see anything useful; to all appearances the system has hung, although I can ping the IP that I see my router assign to it.

If there's a way to persuade the system to come up with sshd and let me login, I can probably fix things up from here. I've never gotten past this point, so I don't know what the system even looks like.

wez, 2024-06-01:
My workaround for this was to install qemu on the macos side, start an ubuntu vm there, mount the physical linux disk, chroot into it, install sshd, set a password, then reboot with the hdmi unplugged. I'm not currently ssh'd in and running a dist upgrade and will see if that allows me to boot with the hdmi cable attached in the future.

wez, 2024-06-01:
I can confirm that 24.04 LTS resolves this issue. It would be great for the installer image to be updated to this release, as it was a bit of an ordeal to get a usable system!