UbuntuAsahi/ubuntu-asahi #109: zipfile.BadZipFile: File is not a zip file (ubuntu-desktop-23.10-20240209.zip 404?)

xekablu, 2024-04-24:
Hi there. I have seen previous issues raised on this and I believe this is related to a required .zip being 404 during build process. Unsure if i am able to supplement this myself manually.

Relevant portion of installer.log as follows:

> 04-24 17:08 root         INFO     Current OS: Macintosh HD / disk3s3s1
> 04-24 17:08 root         INFO     MSG: Choose what to do:
> 04-24 17:08 root         INFO     MSG:   [1mf[22m: Install an OS into free space[m
> 04-24 17:08 root         INFO     MSG:   [1mr[22m: Resize an existing partition to make space for a new OS[m
> 04-24 17:08 root         INFO     MSG:   [1mq[22m: Quit without doing anything[m
> 04-24 17:08 root         INFO     MSG: [1;37m»[1;36m Action (f): [m
> 04-24 17:08 root         INFO     INPUT: 'f'
> 04-24 17:08 root         INFO     MSG: Choose an OS to install:
> 04-24 17:08 root         INFO     MSG:   [1m1[22m: Ubuntu Desktop 23.10[m
> 04-24 17:08 root         INFO     MSG: [1;37m»[1;36m OS: [m
> 04-24 17:08 root         INFO     INPUT: '1'
> 04-24 17:08 root         INFO     Chosen OS: Ubuntu Desktop 23.10
> 04-24 17:08 root         INFO     OS package URL: https://files.tobhe.de/ubuntu/os/ubuntu-desktop-23.10-20240209.zip
> 04-24 17:08 root         INFO     MSG: Downloading OS package info...
> 04-24 17:08 root         ERROR    Exception caught
> Traceback (most recent call last):
>   File "/private/tmp/asahi-install/main.py", line 1058, in <module>
>     InstallerMain(installer_version).main()
>   File "/private/tmp/asahi-install/main.py", line 866, in main
>     while self.main_loop():
>   File "/private/tmp/asahi-install/main.py", line 1021, in main_loop
>     return self.action_install_into_free(parts_free)
>   File "/private/tmp/asahi-install/main.py", line 292, in action_install_into_free
>     self.osins.load_package()
>   File "/private/tmp/asahi-install/osinstall.py", line 48, in load_package
>     self.pkg = zipfile.ZipFile(self.ucache)
>   File "/private/tmp/asahi-install/Frameworks/Python.framework/Versions/3.9/lib/python3.9/zipfile.py", line 1257, in __init__
>     self._RealGetContents()
>   File "/private/tmp/asahi-install/Frameworks/Python.framework/Versions/3.9/lib/python3.9/zipfile.py", line 1324, in _RealGetContents
>     raise BadZipFile("File is not a zip file")
> zipfile.BadZipFile: File is not a zip file
> 04-24 17:08 root         INFO     MSG: If you need to file a bug report, please attach the log file:
> 04-24 17:08 root         INFO     MSG:   /private/tmp/asahi-install/installer.log

Thanks for any input, hopefully this is helpful!

tobhe, 2024-04-24:
Sorry, should be fixed.

closed 2024-04-25 by xekablu

tobhe, 2024-04-25:
Thank you, also fixed the cert!