UbuntuAsahi/ubuntu-asahi #112: Ubuntu 24.04 LTS Support

pixrl, 2024-04-25:
Would it be possible to support Ubuntu 24.04 LTS?

tobhe, 2024-04-25:
Thanks for the report, we are already working on 24.04 support. It will take a few days to get everything running and tested.

jitunayak, 2024-04-27:
> Thanks for the report, we are already working on 24.04 support. It will take a few days to get everything running and tested.

when new version is available. How could we upgrade from 23.x to 24.x ? Do we need to run the script again in mac and choose the 24.04 version ?

tobhe, 2024-04-27:
The upgrade is not yet ready so currently I don't recommend trying. I'll let you know how to do it when it is safe to do.

tobhe, 2024-04-29:
Hi all,

it looks like we have reached a point where release upgrades are relatively safe and we have resolved all the issues we encountered. Here is a step-by-step guide on how to upgrade.

1. Make sure to update all of your packages via `apt update` and `apt upgrade`
2. You should have a ubuntu-asahi meta package with version 1.20 or higher installed. You can check this with `apt list --installed | grep ubuntu-asahi`. It should say something like `ubuntu-asahi/mantic,mantic,now 1.20+23.10 arm64 [installed]`.
3. Run `do-release-upgrade` to start the release upgrade. Read and follow the instructions carefully.

4. **(Optional)** If do-release-upgrade complains about packages not being up to date and you have libreoffice stuck in held-back you might have to remove libreoffice (`apt remove libreoffice*`) and go back to 3. You can reinstall libreoffice after the upgrade on noble.

For now upgrading is the best way to get to 24.04. We will continue working on fresh installer images.

Have fun!

devnoname120, 2024-05-09:
@tobhe How to upgrade if the `ubuntu-asahi` package is kept back on `1.5`? I'm running Ubuntu 22.04 currently.

tobhe, 2024-05-09:
@devnoname120 might be https://github.com/UbuntuAsahi/ubuntu-asahi/issues/113. Can you try manually running `apt install ubuntu-asahi` and see what that reports?

devnoname120, 2024-05-10:
@tobhe It worked!

-------------------

For future visitors here is what I did:

First I manually upgraded the `ubuntu-asahi` package:

```shell
sudo apt intall ubuntu-asahi
```

Then I upgraded to `Ubuntu 23.10`:

```shell
# Replace `Prompt=lts` with `Prompt=normal` in this file:
sudo vim /etc/update-manager/release-upgrades

# `do-release-upgrade -d` returns this error:
# Upgrades to the development release are only available from the latest supported release.
sudo do-release-upgrade

sudo systemctl enable ssh
sudo reboot
```

And then to `Ubuntu 24.04`:

```shell
# Replace `Prompt=lts` with `Prompt=normal` in this file:
sudo vim /etc/update-manager/release-upgrades

sudo do-release-upgrade -d

# During the release upgrade process a GRUB warning told me that I would need to configure it manually. I'm not entirely sure it's needed but just to be safe I ran:
sudo update-grub

sudo systemctl enable ssh
sudo reboot
```

And then I edited this again for future-proofing:

```shell
# Replace `Prompt=lts` with `Prompt=normal` in this file:
sudo vim /etc/update-manager/release-upgrades
```