UbuntuAsahi/ubuntu-asahi #114: Outdated wireplumber package preinstalled

Grarak, 2024-05-11:
I just installed mantic on my macbook air 13 m2. For some reason it came with an outdated wireplumber version of 0.4.14.
I had to manually install a newer version from https://launchpad.net/~pipewire-debian/+archive/ubuntu/wireplumber-upstream/+packages?field.name_filter=&field.status_filter=&field.series_filter= to get the speakers working

- https://launchpad.net/~pipewire-debian/+archive/ubuntu/wireplumber-upstream/+files/libwireplumber-0.4-0_0.4.17-1~ubuntu23.10_arm64.deb
- https://launchpad.net/~pipewire-debian/+archive/ubuntu/wireplumber-upstream/+files/libwireplumber-0.4-0_0.4.17-1~ubuntu23.10_arm64.deb
- Also had to install liblua5.3-0

tobhe, 2024-05-12:
this is unexpected. you should not have to install packages from random ppas (and I don't recommend doing so). The asahi ppa ships wireplumber - 0.4.17-1-ubuntuasahi~ubuntu23.10.1. Can you share the output of `apt-cache madison wireplumber` on your machine?

tobhe, 2024-05-12:
Thanks for the report! Should be fixed in a bit with the wireplumber package in https://launchpad.net/~ubuntu-asahi/+archive/ubuntu/ubuntu-asahi/

Grarak, 2024-05-12:
That was quick, thanks for the fix!

closed 2024-05-17 by eslerm