UbuntuAsahi/ubuntu-asahi #115: Apple's keyboard and mouse do not work on iMac m1

felipereisdev, 2024-05-13:
Hello. I installed Ubuntu Asahi on my Imac M1 and the Apple mouse and keyboard do not appear to connect through Bluetooth. At Fedora Remix Asahi appears in Bluetooth and works perfectly.

tobhe, 2024-05-13:
Hi @felipereisdev, is this a fresh installation? can you show us the output of `apt list --installed | grep apple-arm`?

felipereisdev, 2024-05-13:
> Hi @felipereisdev, is this a fresh installation? can you show us the output of `apt list --installed | grep apple-arm`?

Yes. 

ubuntu@localhost:~$ apt list --installed | grep apple-arm

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

linux-apple-arm-headers-6.6.0-1004/mantic,mantic,now 6.6.0-1004.4 all [installed,automatic]
linux-apple-arm/mantic,mantic,now 6.6.0.1004.4 arm64 [installed,automatic]
linux-headers-6.6.0-1004-apple-arm/mantic,mantic,now 6.6.0-1004.4 arm64 [installed,automatic]
linux-headers-apple-arm/mantic,mantic,now 6.6.0.1004.4 arm64 [installed,automatic]
linux-image-6.6.0-1004-apple-arm/mantic,mantic,now 6.6.0-1004.4 arm64 [installed,automatic]
linux-image-apple-arm/mantic,mantic,now 6.6.0.1004.4 arm64 [installed,automatic]
linux-modules-6.6.0-1004-apple-arm/mantic,mantic,now 6.6.0-1004.4 arm64 [installed,automatic]
linux-modules-extra-6.6.0-1004-apple-arm/mantic,mantic,now 6.6.0-1004.4 arm64 [installed,automatic]

tobhe, 2024-05-14:
You could try updating to 24.04 using `do-release-upgrade -d`. I think this might be a kernel bug that has been fixed in newer versions. 24.04 ships a 6.8 based kernel which might work better.

felipereisdev, 2024-05-14:
> You could try updating to 24.04 using `do-release-upgrade -d`. I think this might be a kernel bug that has been fixed in newer versions. 24.04 ships a 6.8 based kernel which might work better.

Thanks for your help. Unfortunately, the devices still do not appear in Bluetooth even after the update. Is there anything else I can do?

john-cabaj, 2024-05-14:
Which Fedora Asahi Remix version was working? It sounds like F40, kernel 6.8.9-402 is having bluetooth issues with connecting to some bluetooth devices, but discovery is working. If you're using a previous version, there's possibly a regression in upstream.

john-cabaj, 2024-05-14:
Can you try running `lspci -vvv | grep -i bcm` and share the output?

felipereisdev, 2024-05-14:
> Which Fedora Asahi Remix version was working? It sounds like F40, kernel 6.8.9-402 is having bluetooth issues with connecting to some bluetooth devices, but discovery is working. If you're using a previous version, there's possibly a regression in upstream.

I installed the latest version yesterday.

john-cabaj, 2024-05-15:
@felipereisdev - would you be able to give an updated kernel a try? If so, here's the steps:

1. Download deb packages at https://drive.google.com/file/d/1sl56_i2qLbWUiU3qi3bj39WlT9GesibN/view?usp=sharing
2. Extract deb packages `tar -zxvf packages.tar.gz`
3. Navigate to directory containing deb packages `cd packages/`
4. Install deb packages `apt install ./linux*.deb`

Let me know how that goes

felipereisdev, 2024-05-16:
> @felipereisdev - would you be able to give an updated kernel a try? If so, here's the steps:
> 
>     1. Download deb packages at https://drive.google.com/file/d/1sl56_i2qLbWUiU3qi3bj39WlT9GesibN/view?usp=sharing
> 
>     2. Extract deb packages `tar -zxvf packages.tar.gz`
> 
>     3. Navigate to directory containing deb packages `cd packages/`
> 
>     4. Install deb packages `apt install ./linux*.deb`
> 
>     5. Reboot
> 
> 
> Let me know how that goes

Hi. I did the process above and the problem persists.

[image]

john-cabaj, 2024-05-17:
@felipereisdev, I have some more debs to test here - https://drive.google.com/file/d/1xMj08xUBMm1NERe7ijJRPGP5bK0wBX6f/view?usp=sharing

It's also possible that the bluez version we're shipping was updated in Noble and there's a regression. The team is continuing to look into this.

felipereisdev, 2024-05-17:
> @felipereisdev, I have some more debs to test here - https://drive.google.com/file/d/1xMj08xUBMm1NERe7ijJRPGP5bK0wBX6f/view?usp=sharing
> 
> It's also possible that the bluez version we're shipping was updated in Noble and there's a regression. The team is continuing to look into this.

Thank you for your help. I tried that too and it didn't work. I am sending the installation log below:

[image]

eslerm, 2024-05-19:
This _might_ be related to [CVE-2023-45866](https://github.com/marcnewlin/hi_my_name_is_keyboard).

Please see comment 6 from https://bugs.launchpad.net/ubuntu/+source/bluez/+bug/2045931 to test re-enabling classic bonding.

felipereisdev, 2024-05-20:
> Isso _pode_ estar relacionado ao [CVE-2023-45866](https://github.com/marcnewlin/hi_my_name_is_keyboard).
> 
> Por favor, veja o comentário 6 de https://bugs.launchpad.net/ubuntu/+source/bluez/+bug/2045931 para testar a ligação clássica re-ativante.

Hello. I updated ClassicBondedOnly for false and restarted Bluetooth and continues without the mouse and keyboard.

felipereisdev, 2024-05-20:
An important observation that I forgot to mention. Maybe it's a gnome problem. Asahi Linux Fedora only displays the mouse and keyboard in KDE. In the Fedora gnome version it also does not work. @tobhe @john-cabaj @eslerm

eslerm, 2024-05-21:
Thank you for testing Classic Bonding!

Good observation that Gnome might be affected. fwiw, I can see other bluetooth keyboards with the Ubuntu Asahi 24.04 gnome panel.

Could you try the `bluetoothctl` cli tool?
```
$ bluetoothctl 
Waiting to connect to bluetoothd...[bluetooth]# hci0 new_settings: powered bondable ssp br/edr le secure-conn 
[bluetooth]# Agent registered
[bluetooth]# [CHG] Controller C4:00:11:22:33:44 Pairable: yes
[bluetooth]# scan on
[...]
[bluetooth]# SetDiscoveryFilter success
[bluetooth]# Discovery started
[bluetooth]# [CHG] Controller C4:00:11:22:33:44 Discovering: yes
[bluetooth]# [NEW] Device  4F:FF:EE:DD:CC:BB Apple Keyboard
[...]
[bluetooth]# scan off
[bluetooth]# pair 4F:FF:EE:DD:CC:BB
[bluetooth]# connect 4F:FF:EE:DD:CC:BB
```

felipereisdev, 2024-05-21:
> Thank you for testing Classic Bonding!
>
> Good observation that Gnome might be affected. fwiw, I can see other bluetooth keyboards with the Ubuntu Asahi 24.04 gnome panel.
>
> Could you try the [...]

It worked using bluetoothctl and with the help of another machine to identify the address of the mouse and keyboard.

Thanks.