UbuntuAsahi/ubuntu-asahi #116: Firefox not hardware accelerated (native package)

Consistos, 2024-05-18:
I installed Firefox via a PPA (as I understand snap does not use the system's Mesa (similar to flatpack), but the upstream one, which lacks the AGX driver), but Firefox fails to use hardware acceleration.
`eglinfo -B` 's output:
```
GBM platform:
eglinfo: eglInitialize failed

Wayland platform:
EGL API version: 1.5
EGL vendor string: Mesa Project
EGL version string: 1.5
EGL client APIs: OpenGL OpenGL_ES 
OpenGL core profile vendor: Mesa
OpenGL core profile renderer: Apple M2 (G14G B0)
OpenGL core profile version: 4.6 (Core Profile) Mesa 24.0~pre20240228-0asahi6.23.10
OpenGL core profile shading language version: 4.60
OpenGL compatibility profile vendor: Mesa
OpenGL compatibility profile renderer: Apple M2 (G14G B0)
OpenGL compatibility profile version: 4.6 (Compatibility Profile) Mesa 24.0~pre20240228-0asahi6.23.10
OpenGL compatibility profile shading language version: 4.60
OpenGL ES profile vendor: Mesa
OpenGL ES profile renderer: Apple M2 (G14G B0)
OpenGL ES profile version: OpenGL ES 3.2 Mesa 24.0~pre20240228-0asahi6.23.10
OpenGL ES profile shading language version: OpenGL ES GLSL ES 3.20

X11 platform:
EGL API version: 1.5
EGL vendor string: Mesa Project
EGL version string: 1.5
EGL client APIs: OpenGL OpenGL_ES 
OpenGL core profile vendor: Mesa
OpenGL core profile renderer: Apple M2 (G14G B0)
OpenGL core profile version: 4.6 (Core Profile) Mesa 24.0~pre20240228-0asahi6.23.10
OpenGL core profile shading language version: 4.60
OpenGL compatibility profile vendor: Mesa
OpenGL compatibility profile renderer: Apple M2 (G14G B0)
OpenGL compatibility profile version: 4.6 (Compatibility Profile) Mesa 24.0~pre20240228-0asahi6.23.10
OpenGL compatibility profile shading language version: 4.60
OpenGL ES profile vendor: Mesa
OpenGL ES profile renderer: Apple M2 (G14G B0)
OpenGL ES profile version: OpenGL ES 3.2 Mesa 24.0~pre20240228-0asahi6.23.10
OpenGL ES profile shading language version: OpenGL ES GLSL ES 3.20

Surfaceless platform:
EGL API version: 1.5
EGL vendor string: Mesa Project
EGL version string: 1.5
EGL client APIs: OpenGL OpenGL_ES 
OpenGL core profile vendor: Mesa
OpenGL core profile renderer: Apple M2 (G14G B0)
OpenGL core profile version: 4.6 (Core Profile) Mesa 24.0~pre20240228-0asahi6.23.10
OpenGL core profile shading language version: 4.60
OpenGL compatibility profile vendor: Mesa
OpenGL compatibility profile renderer: Apple M2 (G14G B0)
OpenGL compatibility profile version: 4.6 (Compatibility Profile) Mesa 24.0~pre20240228-0asahi6.23.10
OpenGL compatibility profile shading language version: 4.60
OpenGL ES profile vendor: Mesa
OpenGL ES profile renderer: Apple M2 (G14G B0)
OpenGL ES profile version: OpenGL ES 3.2 Mesa 24.0~pre20240228-0asahi6.23.10
OpenGL ES profile shading language version: OpenGL ES GLSL ES 3.20

Device platform:
Device #0:

Platform Device platform:
EGL API version: 1.5
EGL vendor string: Mesa Project
EGL version string: 1.5
EGL client APIs: OpenGL OpenGL_ES 
OpenGL core profile vendor: Mesa
OpenGL core profile renderer: Apple M2 (G14G B0)
[... 148 lines cut ...]
(#0) Error	glxtest: ManageChildProcess failed
(#5) Error	Failed to create EGLConfig for WebRender!
(#6) Error	Failed to create EGLConfig for WebRender!
(#7) Error	Failed GL context creation for hardware WebRender: true
(#8) Error	Failed to create EGLConfig for WebRender!
(#9) Error	Failed to create EGLConfig for WebRender!
(#10) Error	Failed GL context creation for hardware WebRender: true
(#11) Error	Failed to create EGLConfig for WebRender!
(#12) Error	Failed to create EGLConfig for WebRender!
(#13) Error	Failed GL context creation for hardware WebRender: true
(#14) Error	Failed to create EGLConfig for WebRender!
(#15) Error	Failed to create EGLConfig for WebRender!
(#16) Error	Failed GL context creation for WebRender: 0
(#17) Error	FEATURE_FAILURE_WEBRENDER_INITIALIZE_UNSPECIFIED(Create)
(#18) Error	Failed to connect WebRenderBridgeChild. isParent=true
(#19) Error	Fallback WR to SW-WR
```
I have the following error launching Firefox, which seems related:
```
Wayland Proxy [0xffff96570260] Error: StartProxyServer(): bind() error : Permission non accordée
[46521, Main Thread] WARNING: Unable to query dbus: GDBus.Error:org.freedesktop.DBus.Error.AccessDenied: An AppArmor policy prevents this sender from sending this message to this recipient; type="method_call", sender=":1.236" (uid=1000 pid=46521 comm="/usr/lib/firefox/firefox" label="firefox (enforce)") interface="org.freedesktop.DBus" member="ListNames" error name="(unset)" requested_reply="0" destination="org.freedesktop.DBus" (bus): 'glib warning', file /build/firefox-Jlp3yG/firefox-126.0+build2/toolkit/xre/nsSigHandlers.cpp:187

** (firefox:46521): WARNING **: 20:26:54.831: Unable to query dbus: GDBus.Error:org.freedesktop.DBus.Error.AccessDenied: An AppArmor policy prevents this sender from sending this message to this recipient; type="method_call", sender=":1.236" (uid=1000 pid=46521 comm="/usr/lib/firefox/firefox" label="firefox (enforce)") interface="org.freedesktop.DBus" member="ListNames" error name="(unset)" requested_reply="0" destination="org.freedesktop.DBus" (bus)
```
glxinfo` confirms that KDE is accelerated

tobhe, 2024-05-18:
Thanks for the report! Your assumption about snaps is actually wrong, our firefox snap does indeed support graphics acceleration an ubuntu-asahi. We do this using a asahi-specific channel of the gnome-sdk snap with our mesa backported to core22.  Here is what my firefox reports:

[image]

Of course I think we should make sure that the deb package also works properly for people preferring that option.
I'm not entirely sure what causes your error yet, is the apparmor policy causing the error part of the deb package? Which ppa did you use, I am not aware of an "official" up-to-date firefox ppa? Mozilla has recently announced their own deb mirrors, their official builds might work better.

eslerm, 2024-05-19:
@Consistos are you using debs from the downstream "[Mozilla Team](https://wiki.ubuntu.com/MozillaTeam)" PPA?

If you want debs from upstream Mozilla, there is official [Mozilla's packaging](https://support.mozilla.org/en-US/kb/install-firefox-linux).