UbuntuAsahi/ubuntu-asahi #117: KDE's charge limit does not work

Consistos, 2024-05-18:
Battery charge to 100% despite settting it to 75%.

Consistos, 2024-05-19:
The settings are never saved: going to an other energy config tab and back reset the 2 charge limit settings to 100% and 99%. This only applies to them in the advanced energy config panel.
(commenting instead of edit so no one take the wrong lead)

eslerm, 2024-05-19:
@Consistos do you know if this issue occurs in official Ubuntu releases under amd64?

Consistos, 2024-06-14:
It works after [creating the following udev rule](https://social.treehouse.systems/@AsahiLinux/110560192550506827):
```echo 'SUBSYSTEM=="power_supply", KERNEL=="macsmc-battery", ATTR{charge_control_end_threshold}="80"' | sudo tee /etc/udev/rules.d/10-battery.rules```.

closed 2024-06-14 by Consistos

tobhe, 2024-06-14:
Thank you for debugging it! I'll see if we can add that to one of our packages

mkurz, 2024-06-14:
I am not sure if it is a good idea to put this rule in one of your packages.

tobhe, 2024-06-14:
Thanks @mkurz, replied before I properly read it. I assumed it was a fix for kde and not a hard configured limit :) Anyway, we should still try to find a solution for KDE then.