UbuntuAsahi/ubuntu-asahi #118: Update m1n1 for macOS Sonoma 14.5 compatibility

eslerm, 2024-05-19:
Ubuntu Asahi's `m1n1` package must be upgraded to match Apple's memory layout changes.

Per https://social.treehouse.systems/@AsahiLinux/112449204541186432

> Important PSA: Do NOT update to macOS Sonoma 14.5
>
> Update: This is now resolved. [More details here [social.treehouse.systems]](https://social.treehouse.systems/@AsahiLinux/112454079096607233).
>
> We are receiving reports that video output on some or all machines is broken after this update (possibly only desktops). We don't know yet whether this is an Apple regression or a bug/issue in our code. We are still investigating and will update with further details.
>
> Update 1: We have confirmed that Asahi's recoveryOS instance still works, so this is not another major Apple bug situation, nor is there any danger of your machine in general becoming unbootable/unrecoverable. Users should hold off on updating even if they don't plan to use Asahi for the time being, as we don't know yet how easy it will be to fix the problem once Linux becomes unbootable.
>
> Update 2: We believe this only affects M2 series desktops (Mini/Studio).
>
> Update 3: We have identified the root cause, but we're still tracking down some issues. This was caused by a bug/assumption in our code coupled with Apple changing memory layouts in this update. It will be fixed with an upcoming m1n1 update.
>
> Update 4: Fix is available as a m1n1 stage 1 update. Please run the Asahi Linux installer again (curl https://alx.sh | sh) and choose the 'm' option when prompted to upgrade.

See also followup: https://social.treehouse.systems/@AsahiLinux/112454079096607233

tobhe, 2024-05-19:
Already updated to 1.4.14 in our ppa since yesterday, see https://launchpad.net/~ubuntu-asahi/+archive/ubuntu/ubuntu-asahi

Updating m1n1 stage 1 will require a `curl https://alx.sh | sh` in macos and can't be done from within Linux.

eslerm, 2024-05-19:
Can we make this work with `curl -sL https://ubuntuasahi.org/install | sh` ? (Does it need to be added to our REPO_BSAE files?)

tobhe, 2024-05-19:
It probably does already since we use the latest asahi installer (INSTALLER_BASE=https://cdn.asahilinux.org/installer), but I haven't tested it yet.

devnoname120, 2024-05-22:
Note that there is a [stage 2 mitigation](https://www.undeadly.org/cgi?action=article;sid=20240522045252#:~:text=we've%20also%20released%20a%20new%20version%20of%20the%20%22apple-boot%22%20firmware%20(which%20contains%20a%20%22stage2%22%20m1n1)%20that%20has%20a%20workaround%20for%20the%20bug.%20to%20install%20this%20new%20firmware%20on%20openbsd%207.5%20or%20-current%2C%20you%20can%20do%3A%20%23%20fw_update%20%23%20installboot%20sd0%20) in OpenBSD which — as far as I understand — doesn't require booting up macOS. I don't know whether it's available for Asahi Linux or not.

tobhe, 2024-05-22:
@devnoname120 indeed. We also package the fix in our ppa as `m1n1 1.4.14-1ubuntu0.1~asahi1`. You can use `apt list --installed | grep m1n1` to see if you are up-to-date.

devnoname120, 2024-05-22:
@tobhe Hmm for some reason I'm still on `1.4.11-1ubuntu0` and it's the most recent I see. Any ideas why?

```
ubuntu@ubuntu:~$ sudo apt update
Hit:1 http://ports.ubuntu.com/ubuntu-ports noble InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports noble-updates InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports noble-security InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports noble-backports InRelease
Get:5 https://esm.ubuntu.com/apps/ubuntu noble-apps-security InRelease [7469 B]
Get:6 https://esm.ubuntu.com/apps/ubuntu noble-apps-updates InRelease [7468 B]
Get:7 https://esm.ubuntu.com/infra/ubuntu noble-infra-security InRelease [7462 B]
Get:8 https://esm.ubuntu.com/infra/ubuntu noble-infra-updates InRelease [7461 B]
Fetched 29.9 kB in 0s (64.8 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
2 packages can be upgraded. Run 'apt list --upgradable' to see them.

ubuntu@ubuntu:~$ sudo apt list --upgradable
Listing... Done
python3-distupgrade/noble-updates 1:24.04.18 all [upgradable from: 1:24.04.16]
ubuntu-release-upgrader-core/noble-updates 1:24.04.18 all [upgradable from: 1:24.04.16]

ubuntu@ubuntu:~$ sudo apt upgrade m1n1
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
m1n1 is already the newest version (1.4.11-1ubuntu0).
Calculating upgrade... Done
The following upgrades have been deferred due to phasing:
  python3-distupgrade ubuntu-release-upgrader-core
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
```

tobhe, 2024-05-22:
hmm, looks like the ppa is missing. try `sudo add-apt-repository ppa:ubuntu-asahi/ubuntu-asahi`. This could have been caused by the noble update if you did that early enough.

devnoname120, 2024-05-22:
Here is what I see now. I think that I issued `sudo apt upgrade` after `sudo apt update` but I guess the output got mangled by the interactive prompt about `ubuntu-asahi.sources`?

```shell
ubuntu@ubuntu:~$ sudo add-apt-repository ppa:ubuntu-asahi/ubuntu-asahi
[sudo] password for ubuntu:
Repository: 'Types: deb
URIs: https://ppa.launchpadcontent.net/ubuntu-asahi/ubuntu-asahi/ubuntu/
Suites: noble
Components: main
'
Description:
This PPA contains the current stable Ubuntu Asahi packages that are not part of the official archive.
More info: https://launchpad.net/~ubuntu-asahi/+archive/ubuntu/ubuntu-asahi
Adding repository.
Press [ENTER] to continue or Ctrl-c to cancel.
Hit:1 http://ports.ubuntu.com/ubuntu-ports noble InRelease
Get:2 http://ports.ubuntu.com/ubuntu-ports noble-updates InRelease [89.7 kB]
Get:3 https://ppa.launchpadcontent.net/ubuntu-asahi/ubuntu-asahi/ubuntu noble InRelease [17.8 kB]
Hit:4 http://ports.ubuntu.com/ubuntu-ports noble-security InRelease
Hit:5 http://ports.ubuntu.com/ubuntu-ports noble-backports InRelease
Get:6 https://ppa.launchpadcontent.net/ubuntu-asahi/ubuntu-asahi/ubuntu noble/main arm64 Packages [14.9 kB]
Get:7 https://ppa.launchpadcontent.net/ubuntu-asahi/ubuntu-asahi/ubuntu noble/main Translation-en [5864 B]
Get:8 https://esm.ubuntu.com/apps/ubuntu noble-apps-security InRelease [7469 B]
Get:9 https://esm.ubuntu.com/apps/ubuntu noble-apps-updates InRelease [7468 B]
Get:10 https://esm.ubuntu.com/infra/ubuntu noble-infra-security InRelease [7462 B]
Get:11 https://esm.ubuntu.com/infra/ubuntu noble-infra-updates InRelease [7461 B]
Fetched 158 kB in 1s (192 kB/s)
Reading package lists... Done

ubuntu@ubuntu:~$ sudo apt update
Hit:1 http://ports.ubuntu.com/ubuntu-ports noble InRelease
Setting up libgl1-mesa-dri:arm64 (24.1.0~pre20240228-1asahi2.24.04) ...
Setting up linux-image-6.8.0-1005-apple-arm (6.8.0-1005.6) ...
I: /boot/vmlinuz.old is now a symlink to vmlinuz-6.5.0-1013-apple-arm
  .
  mQINBGUTQa4BEADHiZlEAZST37mMqpPKL3wS9cX1WsOx9X8QQaA2pWnuQpp+EUNn
  gbzIVHbE1hTy528GizMc/H/UF8yGf/UGDPDQqI7iyqStNjv7gdm+9g8v6i49sRJ2
@@ -31,4 +32,3 @@
  eQ==
  =a17G
  -----END PGP PUBLIC KEY BLOCK-----
-Enabled: no

Configuration file '/etc/apt/sources.list.d/ubuntu-asahi.sources'
 ==> Modified (by you or by a script) since installation.
 ==> Package distributor has shipped an updated version.
   What would you like to do about it ?  Your options are:
    Y or I  : install the package maintainer's version
    N or O  : keep your currently-installed version
      D     : show the differences between the versions
      Z     : start a shell to examine the situation
 The default action is to keep your current version.
*** ubuntu-asahi.sources (Y/I/N/O/D/Z) [default=N] ? Y
Installing new version of config file /etc/apt/sources.list.d/ubuntu-asahi.sources ...
Refresh gnome-42-2204 snap to asahi/stable channel
/var/lib/dpkg/info/ubuntu-asahi.postinst: 8: snap: not found
Processing triggers for libc-bin (2.39-0ubuntu8.1) ...
Processing triggers for linux-image-6.8.0-1005-apple-arm (6.8.0-1005.6) ...
/etc/kernel/postinst.d/initramfs-tools:
update-initramfs: Generating /boot/initrd.img-6.8.0-1005-apple-arm
W: Possible missing firmware /lib/firmware/renesas_usb_fw.mem for built-in driver xhci_pci
W: Possible missing firmware /lib/firmware/renesas_usb_fw.mem for built-in driver xhci_pci
flash-kernel: deferring update (trigger activated)
/etc/kernel/postinst.d/zz-flash-kernel:###################################################################################################flash-kernel: deferring update (trigger activated)################################################################################......../etc/kernel/postinst.d/zz-update-grub:
Sourcing file `/etc/default/grub'
Generating grub configuration file ...
Found linux image: /boot/vmlinuz-6.8.0-1005-apple-arm
Found initrd image: /boot/initrd.img-6.8.0-1005-apple-arm
Found linux image: /boot/vmlinuz-6.5.0-1013-apple-arm
Found initrd image: /boot/initrd.img-6.5.0-1013-apple-arm
[... 81 lines cut ...]
Found linux image: /boot/vmlinuz-6.5.0-1013-apple-arm
Found initrd image: /boot/initrd.img-6.5.0-1013-apple-arm
Found linux image: /boot/vmlinuz-5.19.5-10-asahi
Found initrd image: /boot/initrd.img-5.19.5-10-asahi
Warning: os-prober will not be executed to detect other bootable partitions.
Systems on them will not be added to the GRUB boot configuration.
Check GRUB_DISABLE_OS_PROBER documentation entry.
Adding boot menu entry for UEFI Firmware Settings ...
done
/etc/kernel/postinst.d/zz-update-m1n1:
/bin/ls: cannot access '/usr/lib/linux-image-*asahi*/apple': No such file or directory
Reading m1n1 config from /etc/m1n1.conf:
Mounted System ESP /dev/nvme0n1p4 at /run/.system-efi
cat: '/*.dtb': No such file or directory
run-parts: /etc/kernel/postinst.d/zz-update-m1n1 exited with return code 1
dpkg: error processing package linux-image-6.8.0-1005-apple-arm (--configure):
 installed linux-image-6.8.0-1005-apple-arm package post-installation script subprocess returned error exit status 1
Processing triggers for flash-kernel (3.107ubuntu8) ...
flash-kernel: requested kernel 6.8.0-1005-apple-arm is in state half-configured.
flash-kernel: quitting cleanly, anticipating a future run
Errors were encountered while processing:
 linux-image-6.8.0-1005-apple-arm
needrestart is being skipped since dpkg has failed
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

tobhe, 2024-05-22:
This is a side effect of your ppa going MIA. You will have to force uninstall `asahi-scripts` and then reinstall it from the new ppa.

What happened is: your ppa got removed because do-release-upgrade without being instructed otherwise simply disables everything but the official archive. The update pulled in the debian version of some of the asahi packages, including asahi-scripts. Now asahi-scripts ships an update hook to update the bootloader, but the device tree path differs from the one in Ubuntu so it can't find them and fails.
We need to find a way to force upgrade your asahi-scripts without running the postinst script.

devnoname120, 2024-05-22:
>EDIT: The best way to fix it manually is probably apt install ubuntu-asahi in the first step to get the ppa pinned correctly, then force reinstall asahi-scripts and then update everything

I tried your solution, see below for the output. Note that I ran `sudo apt reinstall asahi-scripts` but the command prompt was eaten up by output (again).

Should I tried your `EDIT2`?

```
ubuntu@ubuntu:~$ sudo apt install ubuntu-asahi
[sudo] password for ubuntu:
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 2 not upgraded.
1 not fully installed or removed.
Need to get 9224 B of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://ports.ubuntu.com/ubuntu-ports noble/universe arm64 asahi-scripts all 20231219.1-1 [9224 B]
Fetched 9224 B in 0s (121 kB/s)
(Reading database ... 194371 files and directories currently installed.)
Preparing to unpack .../asahi-scripts_20231219.1-1_all.deb ...
Unpacking asahi-scripts (20231219.1-1) over (20231219.1-1) ...
Setting up asahi-scripts (20231219.1-1) ...
update-initramfs: Generating /boot/initrd.img-6.8.0-1005-apple-arm
W: Possible missing firmware /lib/firmware/renesas_usb_fw.mem for built-in driver xhci_pci
W: Possible missing firmware /lib/firmware/renesas_usb_fw.mem for built-in driver xhci_pci
flash-kernel: deferring update (trigger activated)
update-initramfs: Generating /boot/initrd.img-6.5.0-1013-apple-arm
flash-kernel: deferring update (trigger activated)
update-initramfs: Generating /boot/initrd.img-5.19.5-10-asahi
flash-kernel: deferring update (trigger activated)
Setting up linux-image-6.8.0-1005-apple-arm (6.8.0-1005.6) ...
Processing triggers for flash-kernel (3.107ubuntu8) ...
flash-kernel: requested kernel 6.8.0-1005-apple-arm is in state triggers-pending.
flash-kernel: quitting cleanly, anticipating a future run
Processing triggers for linux-image-6.8.0-1005-apple-arm (6.8.0-1005.6) ...
/etc/kernel/postinst.d/initramfs-tools:
update-initramfs: Generating /boot/initrd.img-6.8.0-1005-apple-arm
W: Possible missing firmware /lib/firmware/renesas_usb_fw.mem for built-in driver xhci_pci
W: Possible missing firmware /lib/firmware/renesas_usb_fw.mem for built-in driver xhci_pci
flash-kernel: deferring update (trigger activated)
/etc/kernel/postinst.d/zz-flash-kernel:
flash-kernel: deferring update (trigger activated)
/etc/kernel/postinst.d/zz-update-grub:
Sourcing file `/etc/default/grub'
Generating grub configuration file ...
Found linux image: /boot/vmlinuz-6.8.0-1005-apple-arm
Found initrd image: /boot/initrd.img-6.8.0-1005-apple-arm
Found linux image: /boot/vmlinuz-6.5.0-1013-apple-arm
Found initrd image: /boot/initrd.img-6.5.0-1013-apple-arm
Found linux image: /boot/vmlinuz-5.19.5-10-asahi
Found initrd image: /boot/initrd.img-5.19.5-10-asahi
Warning: os-prober will not be executed to detect other bootable partitions.
Systems on them will not be added to the GRUB boot configuration.
Check GRUB_DISABLE_OS_PROBER documentation entry.
Adding boot menu entry for UEFI Firmware Settings ...
done
/etc/kernel/postinst.d/zz-update-m1n1:
/bin/ls: cannot access '/usr/lib/linux-image-*asahi*/apple': No such file or directory
Reading m1n1 config from /etc/m1n1.conf:
Mounted System ESP /dev/nvme0n1p4 at /run/.system-efi
cat: '/*.dtb': No such file or directory
run-parts: /etc/kernel/postinst.d/zz-update-m1n1 exited with return code 1
dpkg: error processing package linux-image-6.8.0-1005-apple-arm (--configure):
 installed linux-image-6.8.0-1005-apple-arm package post-installation script subprocess returned error exit status 1
Processing triggers for flash-kernel (3.107ubuntu8) ...
flash-kernel: requested kernel 6.8.0-1005-apple-arm is in state half-configured.
flash-kernel: quitting cleanly, anticipating a future run
Errors were encountered while processing:
 linux-image-6.8.0-1005-apple-arm
needrestart is being skipped since dpkg has failed
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

tobhe, 2024-05-22:
Looks good! You can remove `/etc/apt/sources.list.d/ubuntu-asahi-ubuntu-ubuntu-asahi-noble.sources` now because the ubuntu-asahi meta package added that a second time and you want to keep the one from the package. You can manually run update-m1n1 once more before rebooting just to be sure.

Sorry for the trouble! I wonder if we might need to do an info post for others. We fixed this in for most people with an ubuntu-asahi update but if people upgraded before that they might have hit the same issue.

devnoname120, 2024-05-22:
```shell
ubuntu@ubuntu:~$ sudo rm /etc/apt/sources.list.d/ubuntu-asahi-ubuntu-ubuntu-asahi-noble.sources
[sudo] password for ubuntu:

```

```shell
ubuntu@ubuntu:~$ sudo apt update
Hit:1 http://ports.ubuntu.com/ubuntu-ports noble InRelease
Get:2 http://ports.ubuntu.com/ubuntu-ports noble-updates InRelease [89.7 kB]
Hit:3 https://ppa.launchpadcontent.net/ubuntu-asahi/ubuntu-asahi/ubuntu noble InRelease
Get:4 https://esm.ubuntu.com/apps/ubuntu noble-apps-security InRelease [7469 B]
Hit:5 http://ports.ubuntu.com/ubuntu-ports noble-security InRelease
Get:6 https://esm.ubuntu.com/apps/ubuntu noble-apps-updates InRelease [7468 B]
Hit:7 http://ports.ubuntu.com/ubuntu-ports noble-backports InRelease
Get:8 https://esm.ubuntu.com/infra/ubuntu noble-infra-security InRelease [7462 B]
Get:9 https://esm.ubuntu.com/infra/ubuntu noble-infra-updates InRelease [7461 B]
Fetched 120 kB in 1s (199 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
2 packages can be upgraded. Run 'apt list --upgradable' to see them.
```

```shell
ubuntu@ubuntu:~$ sudo update-m1n1
/bin/ls: cannot access '/usr/lib/linux-image-*asahi*/apple': No such file or directory
Reading m1n1 config from /etc/m1n1.conf:
Mounted System ESP /dev/nvme0n1p4 at /run/.system-efi
cat: '/*.dtb': No such file or directory
```

Does this look good?

tobhe, 2024-05-22:
Oh no, that still seems to be the wrong asahi-scripts version causing the same bug you saw previously...
what does `apt list --upgradable` say? 

On my machine I'm seeing:
```
user@ubuntu:~$ apt list --installed | grep asahi-scripts
asahi-scripts/now 20231219.1-1ubuntu1 all [installed,local]

```

devnoname120, 2024-05-22:
```shell
ubuntu@ubuntu:~$ apt list --upgradable
Listing... Done
python3-distupgrade/noble-updates 1:24.04.18 all [upgradable from: 1:24.04.16]
ubuntu-release-upgrader-core/noble-updates 1:24.04.18 all [upgradable from: 1:24.04.16]
```

tobhe, 2024-05-22:
Ok, so this seems to be my mistake. Looks like we missed this when moving everything to the new ppa for noble. Should be there in a few minutes.

tobhe, 2024-05-22:
Should be available now

devnoname120, 2024-05-22:
```shell
ubuntu@ubuntu:~$ sudo apt update
[sudo] password for ubuntu:
Hit:1 http://ports.ubuntu.com/ubuntu-ports noble InRelease
Get:2 http://ports.ubuntu.com/ubuntu-ports noble-updates InRelease [89.7 kB]
Get:3 https://esm.ubuntu.com/apps/ubuntu noble-apps-security InRelease [7469 B]
Get:4 https://esm.ubuntu.com/apps/ubuntu noble-apps-updates InRelease [7468 B]
Get:5 http://ports.ubuntu.com/ubuntu-ports noble-security InRelease [89.7 kB]
Get:6 https://esm.ubuntu.com/infra/ubuntu noble-infra-security InRelease [7462 B]
Get:7 https://esm.ubuntu.com/infra/ubuntu noble-infra-updates InRelease [7461 B]
Hit:8 http://ports.ubuntu.com/ubuntu-ports noble-backports InRelease
Get:9 http://ports.ubuntu.com/ubuntu-ports noble-updates/main arm64 Packages [34.7 kB]
Get:10 http://ports.ubuntu.com/ubuntu-ports noble-updates/universe arm64 Packages [22.4 kB]
Get:11 https://ppa.launchpadcontent.net/ubuntu-asahi/ubuntu-asahi/ubuntu noble InRelease [17.8 kB]
Get:12 https://ppa.launchpadcontent.net/ubuntu-asahi/ubuntu-asahi/ubuntu noble/main arm64 Packages [15.1 kB]
Get:13 https://ppa.launchpadcontent.net/ubuntu-asahi/ubuntu-asahi/ubuntu noble/main Translation-en [5940 B]
Fetched 305 kB in 2s (163 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
3 packages can be upgraded. Run 'apt list --upgradable' to see them.
```

```shell
ubuntu@ubuntu:~$ sudo apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  asahi-fwextract liblzfse1 linux-image-5.19.5-10-asahi linux-modules-5.19.5-10-asahi
Use 'sudo apt autoremove' to remove them.
The following upgrades have been deferred due to phasing:
  python3-distupgrade ubuntu-release-upgrader-core
The following packages will be upgraded:
  asahi-scripts
1 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Need to get 9408 B of archives.
After this operation, 1024 B of additional disk space will be used.
Do you want to continue? [Y/n]
Get:1 https://ppa.launchpadcontent.net/ubuntu-asahi/ubuntu-asahi/ubuntu noble/main arm64 asahi-scripts all 20231219.1-1ubuntu1 [9408 B]
Fetched 9408 B in 2s (3783 B/s)
(Reading database ... 194371 files and directories currently installed.)
Preparing to unpack .../asahi-scripts_20231219.1-1ubuntu1_all.deb ...
No diversion 'diversion of /usr/sbin/update-grub by asahi-scripts', none removed.
Unpacking asahi-scripts (20231219.1-1ubuntu1) over (20231219.1-1) ...
Setting up asahi-scripts (20231219.1-1ubuntu1) ...
update-initramfs: Generating /boot/initrd.img-6.8.0-1005-apple-arm
flash-kernel: deferring update (trigger activated)
update-initramfs: Generating /boot/initrd.img-6.5.0-1013-apple-arm
flash-kernel: deferring update (trigger activated)
update-initramfs: Generating /boot/initrd.img-5.19.5-10-asahi
flash-kernel: deferring update (trigger activated)
Processing triggers for flash-kernel (3.107ubuntu8) ...
System running in EFI mode, skipping.....
Scanning processes...
Scanning processor microcode...
Scanning linux images...

Pending kernel upgrade!
Running kernel version:
  6.5.0-1013-apple-arm
Diagnostics:
  The currently running kernel version is not the expected kernel version 6.8.0-1005-apple-arm.

Restarting the system to load the new kernel will not be handled automatically, so you should consider rebooting.

The processor microcode seems to be up-to-date.

No services need to be restarted.
[... 83 lines cut ...]
flash-kernel: Kernel 5.19.5-10-asahi has been removed.
flash-kernel: A higher version (6.8.0-1005-apple-arm) is still installed, no reflashing required.
/etc/kernel/postrm.d/zz-update-grub:
Sourcing file `/etc/default/grub'
Generating grub configuration file ...
Found linux image: /boot/vmlinuz-6.8.0-1005-apple-arm
Found initrd image: /boot/initrd.img-6.8.0-1005-apple-arm
Found linux image: /boot/vmlinuz-6.5.0-1013-apple-arm
Found initrd image: /boot/initrd.img-6.5.0-1013-apple-arm
Warning: os-prober will not be executed to detect other bootable partitions.
Systems on them will not be added to the GRUB boot configuration.
Check GRUB_DISABLE_OS_PROBER documentation entry.
Adding boot menu entry for UEFI Firmware Settings ...
done
/etc/kernel/postrm.d/zz-update-m1n1:
Reading m1n1 config from /etc/m1n1.conf:
Mounted System ESP /dev/nvme0n1p4 at /run/.system-efi
m1n1 updated at /run/.system-efi/m1n1/boot.bin
Removing linux-modules-5.19.5-10-asahi (5.19.5-10) ...
Processing triggers for libc-bin (2.39-0ubuntu8.1) ...
```

And I then rebooted and everything looks fine 👌

I will update macOS and reboot to Ubuntu to see if the mitigation works. And then I'll update the stage 1 bootloader from the macOS recovery.

tobhe, 2024-05-22:
Awesome, thanks for your patience :) One less bug.

devnoname120, 2024-05-22:
Here is what I did to upgrade and test:
- Booted to macOS.
- Upgraded macOS Sonoma from `14.4.1` to `14.5`.
- Rebooted, and it automatically booted to macOS.
- Turned off my Mac.
- Long-pressed the power button, and opened Ubuntu Asahi from the boot selection screen.
- Confirmed that Ubuntu Asahi booted up without issues and was working properly.
- `sudo poweroff`.

Here is what I did for upgrading m1n1:
- Short-pressed the power button, the Mac automatically booted to macOS without input from me.
- Ran `curl -sL https://ubuntuasahi.org/install | sh`  (so I confirm it works 🎉, no need to run `curl https://alx.sh/ | sh`).
- Pressed the `m` key in order to `Upgrade m1n1 on an existing OS`.
- It informed me that it would upgrade the `m1n1 stage 1` version from `v1.4.11` to `v1.4.14`.
- I followed the rest of the instructions, and the script eventually turned off my Mac.
- Long-pressed the power button, and opened Ubuntu Asahi from the boot selection screen.
- It opened some kind of macOS recovery with a console and I followed the instructions from there.
- The Mac automatically rebooted, this time to Ubuntu Asahi.
- Confirmed that Ubuntu Asahi booted up without issues and was working properly.
- Ran `sudo reboot`, and Ubuntu Asahi booted up (so it's the default again).
- Ran `sudo poweroff` and then short-pressed the power button, and Ubuntu Asahi booted up (just a sanity check).

-----------------

So everything works and is solved on my side. 🎉

@tobhe No, thank _you_ for your patience. I'm incredibly grateful for your work! 🙏

devnoname120, 2024-05-22:
@tobhe I owe you a beer next time I pass by München! 😄

tobhe, 2024-05-22:
@devnoname120 that is an offer I can hardly say no to :) feel free to ping me when you are around