UbuntuAsahi/ubuntu-asahi #119: Kernel Panic on Upgrade from 23.10 to 24.04 on M1 Max Mac Studio

umnaut, 2024-05-20:
I'm experiencing a kernel panic issue when attempting to upgrade from Ubuntu 23.10 to 24.04 on my M1 Max Mac Studio. The initial 23.10 install was done yesterday, so m1n1 has always been at v1.4.14 for this system.

## Steps to Reproduce:

 - Updated to macOS 14.5.
- Followed the ubuntu-asahi Linux installation process.
- Installed Ubuntu 23.10 successfully.
- Upgrade to Ubuntu 24.04
- Reboot
## Issue:

The system experiences a kernel panic after GRUB boots the upgraded system.
I have wiped and retried the installation process three times with the same results.

Here’s a screenshot of the kernel panic: 

[image]

## Additional  Information:
The system eventually booted after looping for over an hour. From the now-booted system: 
`m1n1/noble, now 1.4.14-1ubuntu0.1~asahi1 arm64`

### Logs:

Log from successful boot: [Journalctl Log](https://pastebin.com/LdXXCKb0)
Previous, failed boot logs:
    [Part 1](https://pastebin.com/Ywxh6hb9)
    [Part 2](https://pastebin.com/wiPet75D)

Please let me know if any additional information is required.

Thank you!

john-cabaj, 2024-05-21:
@umnaut is this now booting again consistently? Or does it loop every boot and take 20 minutes?

umnaut, 2024-05-21:
I sent the reboot command, started a timer, and observed the following:

- 0 to 4 mins:  During several boots, the system displayed a black screen with a single blinking underscore character after m1n1. Each boot lasted several seconds before rebooting again.
- 4 mins 10 secs: On a single boot, the Grub menu appeared after m1n1. Following the Grub timeout, the Ubuntu greeter showed up. After a few seconds at the greeter, the system rebooted to the Boot Recovery Assistant.

I had initially left the system on after the single successful boot to be able to run diagnostic commands, but I've now rebooted to gather this information.

umnaut, 2024-05-21:
I attempted to boot Ubuntu from the Boot Recovery Assistant and observed a similar pattern:

    0:50 - U-Boot -> underscore
    1:25 - U-Boot -> underscore
    2:10 - U-Boot -> Grub -> underscore
    3:30 - U-Boot -> Grub -> underscore
    4:40 - U-Boot -> kernel panic
    5:50 - Boot Recovery Assistant

t-komatu, 2024-05-23:
I had the same problem on mac studio m2ultra.
It seems that some information remains when rebooting from macosx.
If possible, try rebooting, force quitting, or shutting down from grub and waiting a few minutes before turning the power back on.

t-komatu, 2024-05-23:
Sorry, please don't reboot

marcan, 2024-12-15:
This is a known hardware bug in the sdhci chipset which causes errors when autosuspend triggers before device probe. You need [this](https://pagure.io/fedora-asahi/fedora-asahi-remix-scripts/blob/main/f/65-autosuspend-override-asahi.hwdb) workaround, I don't know if ubuntu-asahi ships it? cc @tobhe (be aware of the dependencies, this needs to be installed on the system, then hwdb updated, then the initramfs built with that hwdb so it picks it up too).

tobhe, 2024-12-15:
Thanks @marcan! we don't currently ship it because I wasn't aware of that. I'll make sure to change that quickly :)