UbuntuAsahi/ubuntu-asahi #120: Ubuntu fails to update

roloit, 2024-05-22:
Hi. So I downloaded Asahi Ubuntu 23.10 on my M2 mac, and I am getting an error when updating. 
Here's the issue: `E: Failed to fetch http://ports.ubuntu.com/ubuntu/mantic/dists/main/restricted/source/Sources  404  Not Found [IP: 185.125.190.36 80]
E: Some index files failed to download. They have been ignored, or old ones used instead.
`
I tried to edit the 
`/etc/apt/sources.list` file to change the mirror, but it still doesn't work.
It seems like http://ports.ubuntu.com is "not found", however when I open it in a browser, it works.
Because of that, I also can't upgrade to the 24.04 version.
Please check if there's a solution to this.

tobhe, 2024-05-22:
Hi @roloit, can you please share the full content of your `/etc/apt/sources.list` file? I would expect the path to be `http://ports.ubuntu.com/ubuntu-ports/` and not `http://ports.ubuntu.com/ubuntu/`.

roloit, 2024-05-22:
Hello, here: `deb [trusted=yes] http://ports.ubuntu.com/ubuntu-ports/ mantic main restricted
deb-src [trusted=yes] http://ports.ubuntu.com/ubuntu-ports/ mantic main restricted`
I changed it to ubuntu-ports, but it still seems to fail.
`~$ sudo apt-get update
Get:1 http://ports.ubuntu.com/ubuntu-ports mantic InRelease [256 kB]
Get:2 http://ports.ubuntu.com/ubuntu-ports mantic/main Sources [1,390 kB]
Get:3 http://ports.ubuntu.com/ubuntu-ports mantic/restricted Sources [24.7 kB]
Get:4 http://ports.ubuntu.com/ubuntu-ports mantic/main arm64 Packages [1,384 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports mantic/main i386 Packages [1,040 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports mantic/main i386 Packages [1,040 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports mantic/main i386 Packages [1,040 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports mantic/main i386 Packages [1,040 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports mantic/main i386 Packages [1,040 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports mantic/main i386 Packages [1,040 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports mantic/main i386 Packages [1,040 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports mantic/main i386 Packages [1,040 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports mantic/main i386 Packages [1,040 kB]
Ign:5 http://ports.ubuntu.com/ubuntu-ports mantic/main i386 Packages
Ign:15 http://ports.ubuntu.com/ubuntu-ports mantic/restricted i386 Packages
Ign:5 http://ports.ubuntu.com/ubuntu-ports mantic/main i386 Packages
Ign:15 http://ports.ubuntu.com/ubuntu-ports mantic/restricted i386 Packages
Ign:5 http://ports.ubuntu.com/ubuntu-ports mantic/main i386 Packages
Ign:15 http://ports.ubuntu.com/ubuntu-ports mantic/restricted i386 Packages
Ign:5 http://ports.ubuntu.com/ubuntu-ports mantic/main i386 Packages
Ign:15 http://ports.ubuntu.com/ubuntu-ports mantic/restricted i386 Packages
Ign:5 http://ports.ubuntu.com/ubuntu-ports mantic/main i386 Packages
Ign:15 http://ports.ubuntu.com/ubuntu-ports mantic/restricted i386 Packages
Err:5 http://ports.ubuntu.com/ubuntu-ports mantic/main i386 Packages
  404  Not Found [IP: 185.125.190.36 80]
Ign:15 http://ports.ubuntu.com/ubuntu-ports mantic/restricted i386 Packages
Fetched 256 kB in 1s (188 kB/s)
Reading package lists... Done
E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/dists/mantic/main/binary-i386/Packages  404  Not Found [IP: 185.125.190.36 80]
E: Some index files failed to download. They have been ignored, or old ones used instead.
`

tobhe, 2024-05-22:
@roloit it looks like you manually changed that file before and made a mistake. There should not be any i386 lines in there.

Below is the a fresh sources.list extracted from the last image. Restoring that probably fixes your problem.

```
# See http://help.ubuntu.com/community/UpgradeNotes for how to upgrade to
# newer versions of the distribution.
deb http://ports.ubuntu.com/ubuntu-ports/ mantic main restricted
# deb-src http://ports.ubuntu.com/ubuntu-ports/ mantic main restricted

## Major bug fix updates produced after the final release of the
## distribution.
deb http://ports.ubuntu.com/ubuntu-ports/ mantic-updates main restricted
# deb-src http://ports.ubuntu.com/ubuntu-ports/ mantic-updates main restricted

## N.B. software from this repository is ENTIRELY UNSUPPORTED by the Ubuntu
## team. Also, please note that software in universe WILL NOT receive any
## review or updates from the Ubuntu security team.
deb http://ports.ubuntu.com/ubuntu-ports/ mantic universe
# deb-src http://ports.ubuntu.com/ubuntu-ports/ mantic universe
deb http://ports.ubuntu.com/ubuntu-ports/ mantic-updates universe
# deb-src http://ports.ubuntu.com/ubuntu-ports/ mantic-updates universe

## N.B. software from this repository is ENTIRELY UNSUPPORTED by the Ubuntu
## team, and may not be under a free licence. Please satisfy yourself as to
## your rights to use the software. Also, please note that software in
## multiverse WILL NOT receive any review or updates from the Ubuntu
## security team.
deb http://ports.ubuntu.com/ubuntu-ports/ mantic multiverse
# deb-src http://ports.ubuntu.com/ubuntu-ports/ mantic multiverse
deb http://ports.ubuntu.com/ubuntu-ports/ mantic-updates multiverse
# deb-src http://ports.ubuntu.com/ubuntu-ports/ mantic-updates multiverse

## N.B. software from this repository may not have been tested as
## extensively as that contained in the main release, although it includes
## newer versions of some applications which may provide useful features.
## Also, please note that software in backports WILL NOT receive any review
## or updates from the Ubuntu security team.
deb http://ports.ubuntu.com/ubuntu-ports/ mantic-backports main restricted universe multiverse
# deb-src http://ports.ubuntu.com/ubuntu-ports/ mantic-backports main restricted universe multiverse

deb http://ports.ubuntu.com/ubuntu-ports/ mantic-security main restricted
# deb-src http://ports.ubuntu.com/ubuntu-ports/ mantic-security main restricted
deb http://ports.ubuntu.com/ubuntu-ports/ mantic-security universe
# deb-src http://ports.ubuntu.com/ubuntu-ports/ mantic-security universe
deb http://ports.ubuntu.com/ubuntu-ports/ mantic-security multiverse
# deb-src http://ports.ubuntu.com/ubuntu-ports/ mantic-security multiverse
```

roloit, 2024-05-22:
Hi again. So I replaced the whole `sources.list` file with the default code that you provided, and, sorry to disappoint, but it didn't fix it... Still get the same errors with the `i386` packages...

roloit, 2024-05-22:
Hey, good news here!!! I finally fixed the issue, by running the command 
`sudo dpkg --remove-architecture i386`
Don't know how it got installed there. Have a good day!