UbuntuAsahi/ubuntu-asahi #124: external display not working

KenWuqianghao, 2024-06-18:
external display thru type c and hdmi doesn't seem to be working, any pointers on that?

tobhe, 2024-06-18:
Hey @KenWuqianghao, display over USB-C is not yet working, this is a known limitation. It is listed as "DP Alt Mode" here in the [Asahi feature support table](https://github.com/AsahiLinux/docs/wiki/M2-Series-Feature-Support).

LazerKnown, 2025-06-24:
so, if i use hdmi on a usb-c dongle, will it work??? is there any way to yse an external display?

tobhe, 2025-06-24:
No, usb-c to HDMI will not work. The only thing that works is the actual HDMI port on some pro laptops (this is not a Ubuntu limitation but true for all Asahi variants).

LazerKnown, 2025-06-24:
im sad, im using this os on macbook air m2, IT DOES NOT HAVE ANY HDMI PORT