UbuntuAsahi/ubuntu-asahi #125: Speakers not detected (No sound)

ansfidine, 2024-06-28:
No output and input device detected on my MacBook Pro M1 pro
++Ubuntu 24.04 LTS

tobhe, 2024-06-28:
Is this a fresh installation? Does updating packages and rebooting help?

ansfidine, 2024-06-28:
> Is this a fresh installation? Does updating packages and rebooting help? 

I've done it and still the same thing so I've switched to fedora asahi and it's working well after tried all solutions I could find.

tobhe, 2024-07-05:
Thanks @Talme, this adds some valuable new info. Could you try `systemctl disable speakersafetyd.service` and reboot to see if that helps? Speakersafetyd logs and possibly related error messages in dmesg would be helpful.

Talme, 2024-07-05:
No luck after reboot and disabling speakersafetyd.service

```
$ journalctl -u speakersafetyd
-- Boot a5abb8c80daf45049e7fab05c64b057c --
Jul 05 16:02:30 kuuki systemd[1]: Started speakersafetyd.service - Speaker Protection Daemon.
Jul 05 16:02:30 kuuki speakersafetyd[879]: INFO  [speakersafetyd] Starting up
Jul 05 16:02:30 kuuki speakersafetyd[879]: INFO  [speakersafetyd] Config base: "/usr/share/speakersafetyd/"
Jul 05 16:02:30 kuuki speakersafetyd[879]: INFO  [speakersafetyd] Machine: apple,j313
Jul 05 16:02:30 kuuki speakersafetyd[879]: INFO  [speakersafetyd] Config file: "/usr/share/speakersafetyd/apple/j313.conf"
Jul 05 16:02:30 kuuki speakersafetyd[879]: INFO  [speakersafetyd] Device: hw:AppleJ313
Jul 05 16:02:30 kuuki speakersafetyd[879]: INFO  [speakersafetyd::uclamp] Set task uclamp to 0:64
Jul 05 16:02:30 kuuki speakersafetyd[879]: INFO  [speakersafetyd] Enabling blackbox, path: "/var/lib/speakersafetyd/blackbox"
Jul 05 16:02:30 kuuki speakersafetyd[879]: INFO  [speakersafetyd] Found 2 speakers
Jul 05 16:02:30 kuuki speakersafetyd[879]: INFO  [speakersafetyd] Opening control device
Jul 05 16:02:30 kuuki speakersafetyd[879]: INFO  [speakersafetyd] Startup mode: Cold boot
Jul 05 16:02:30 kuuki speakersafetyd[879]: INFO  [speakersafetyd::types] Speaker [Left]:
Jul 05 16:02:30 kuuki speakersafetyd[879]: INFO  [speakersafetyd::types]   Group: 0
Jul 05 16:02:30 kuuki speakersafetyd[879]: INFO  [speakersafetyd::types]   Max temperature: 120.0 °C
Jul 05 16:02:30 kuuki speakersafetyd[879]: INFO  [speakersafetyd::types]   Amp gain: 16 dBV
Jul 05 16:02:30 kuuki speakersafetyd[879]: INFO  [speakersafetyd::types]   Max power: 1.08 W
Jul 05 16:02:30 kuuki speakersafetyd[879]: INFO  [speakersafetyd::types]   Peak power: 16.249273 W
Jul 05 16:02:30 kuuki speakersafetyd[879]: INFO  [speakersafetyd::types]   Min gain: -11.79 dB
Jul 05 16:02:30 kuuki speakersafetyd[879]: INFO  [speakersafetyd::types] Speaker [Right]:
Jul 05 16:02:30 kuuki speakersafetyd[879]: INFO  [speakersafetyd::types]   Group: 0
Jul 05 16:02:30 kuuki speakersafetyd[879]: INFO  [speakersafetyd::types]   Max temperature: 120.0 °C
Jul 05 16:02:30 kuuki speakersafetyd[879]: INFO  [speakersafetyd::types]   Amp gain: 16 dBV
Jul 05 16:02:30 kuuki speakersafetyd[879]: INFO  [speakersafetyd::types]   Max power: 1.08 W
Jul 05 16:02:30 kuuki speakersafetyd[879]: INFO  [speakersafetyd::types]   Peak power: 16.249273 W
Jul 05 16:02:30 kuuki speakersafetyd[879]: INFO  [speakersafetyd::types]   Min gain: -11.79 dB
Jul 05 16:02:30 kuuki speakersafetyd[879]: PCM rate: 8000..192000
```

tobhe, 2024-07-05:
Ok, so this doesn't look to bad. `INFO  [speakersafetyd] Found 2 speakers` sounds like the speakers are there and found by speakersafetyd. So our problem seems to happen somewhere between speakersafetyd and pipewire...
Maybe the update pulled a new pipewire or wireplumber version that is compiled with the wrong feature set. Can you check your apt log to see which packages were updated?

Talme, 2024-07-05:
Indeed there were pipewire updates (see row 13 in attached file
[history.log](https://github.com/user-attachments/files/16110901/history.log) )

Incidentally, playing sound directly to a device with `aplay -D plughw:CARD=AppleJ313,DEV=1 /usr/share/sounds/alsa/Front_Center.wav` actually plays sound on the speaker. Would it mean that there is some device selection that broke somewhere?

tobhe, 2024-07-05:
It is complicated. We enforce a DSP chain to process the raw audio and make it sound properly using pipewire and a few plugins, it looks like something in that chain might be broken.
Can you show what `apt list --installed | grep pipewire` says and maybe `apt policy`. Normally our ppa should be pinned with a higher priority than the regular archive to prefer our patched packages. It looks like you might have gotten the regular archive pipewire instead.

Talme, 2024-07-05:
Sure!

apt list --installed | grep pipewire:
```
gstreamer1.0-pipewire/noble-updates,now 1.0.5-1ubuntu1 arm64 [installed,automatic]
libpipewire-0.3-0t64/noble-updates,now 1.0.5-1ubuntu1 arm64 [installed,automatic]
libpipewire-0.3-common/noble-updates,now 1.0.5-1ubuntu1 all [installed,automatic]
libpipewire-0.3-modules/noble-updates,now 1.0.5-1ubuntu1 arm64 [installed,automatic]
pipewire-alsa/noble-updates,now 1.0.5-1ubuntu1 arm64 [installed,automatic]
pipewire-audio/noble-updates,now 1.0.5-1ubuntu1 all [installed,automatic]
pipewire-bin/noble-updates,now 1.0.5-1ubuntu1 arm64 [installed,automatic]
pipewire-pulse/noble-updates,now 1.0.5-1ubuntu1 arm64 [installed,automatic]
pipewire/noble-updates,now 1.0.5-1ubuntu1 arm64 [installed]
```

apt policy:
```
Package files:
 100 /var/lib/dpkg/status
     release a=now
 500 http://ports.ubuntu.com/ubuntu-ports noble-security/multiverse arm64 Packages
     release v=24.04,o=Ubuntu,a=noble-security,n=noble,l=Ubuntu,c=multiverse,b=arm64
     origin ports.ubuntu.com
 500 http://ports.ubuntu.com/ubuntu-ports noble-security/restricted arm64 Packages
     release v=24.04,o=Ubuntu,a=noble-security,n=noble,l=Ubuntu,c=restricted,b=arm64
     origin ports.ubuntu.com
 500 http://ports.ubuntu.com/ubuntu-ports noble-security/universe arm64 Packages
     release v=24.04,o=Ubuntu,a=noble-security,n=noble,l=Ubuntu,c=universe,b=arm64
     origin ports.ubuntu.com
 500 http://ports.ubuntu.com/ubuntu-ports noble-security/main arm64 Packages
     release v=24.04,o=Ubuntu,a=noble-security,n=noble,l=Ubuntu,c=main,b=arm64
     origin ports.ubuntu.com
 100 http://ports.ubuntu.com/ubuntu-ports noble-backports/universe arm64 Packages
     release v=24.04,o=Ubuntu,a=noble-backports,n=noble,l=Ubuntu,c=universe,b=arm64
     origin ports.ubuntu.com
 500 http://ports.ubuntu.com/ubuntu-ports noble-updates/multiverse arm64 Packages
     release v=24.04,o=Ubuntu,a=noble-updates,n=noble,l=Ubuntu,c=multiverse,b=arm64
     origin ports.ubuntu.com
 500 http://ports.ubuntu.com/ubuntu-ports noble-updates/restricted arm64 Packages
     release v=24.04,o=Ubuntu,a=noble-updates,n=noble,l=Ubuntu,c=restricted,b=arm64
     origin ports.ubuntu.com
 500 http://ports.ubuntu.com/ubuntu-ports noble-updates/universe arm64 Packages
     release v=24.04,o=Ubuntu,a=noble-updates,n=noble,l=Ubuntu,c=universe,b=arm64
     origin ports.ubuntu.com
 500 http://ports.ubuntu.com/ubuntu-ports noble-updates/main arm64 Packages
     release v=24.04,o=Ubuntu,a=noble-updates,n=noble,l=Ubuntu,c=main,b=arm64
     origin ports.ubuntu.com
 500 http://ports.ubuntu.com/ubuntu-ports noble/multiverse arm64 Packages
     release v=24.04,o=Ubuntu,a=noble,n=noble,l=Ubuntu,c=multiverse,b=arm64
     origin ports.ubuntu.com
 500 http://ports.ubuntu.com/ubuntu-ports noble/restricted arm64 Packages
     release v=24.04,o=Ubuntu,a=noble,n=noble,l=Ubuntu,c=restricted,b=arm64
     origin ports.ubuntu.com
 500 http://ports.ubuntu.com/ubuntu-ports noble/universe arm64 Packages
     release v=24.04,o=Ubuntu,a=noble,n=noble,l=Ubuntu,c=universe,b=arm64
     origin ports.ubuntu.com
 500 http://ports.ubuntu.com/ubuntu-ports noble/main arm64 Packages
     release v=24.04,o=Ubuntu,a=noble,n=noble,l=Ubuntu,c=main,b=arm64
     origin ports.ubuntu.com
 500 https://ppa.launchpadcontent.net/ubuntu-asahi/ubuntu-asahi/ubuntu noble/main arm64 Packages
     release v=24.04,o=LP-PPA-ubuntu-asahi-ubuntu-asahi,a=noble,n=noble,l=Ubuntu Asahi,c=main,b=arm64
     origin ppa.launchpadcontent.net
 500 http://ppa.launchpad.net/ubuntu-asahi/ubuntu-asahi/ubuntu noble/main arm64 Packages
     release v=24.04,o=LP-PPA-ubuntu-asahi-ubuntu-asahi,a=noble,n=noble,l=Ubuntu Asahi,c=main,b=arm64
     origin ppa.launchpad.net
Pinned packages:
```

tobhe, 2024-07-05:
Ok, I think I might have a fix for you. new version of pipewire is currently building in our testing ppa at https://launchpad.net/~ubuntu-asahi/+archive/ubuntu/ubuntu-asahi-next
Once that is done I can publish it for everyone. If you like you can grab it from there or just wait a bit until you get the fix later today.

Talme, 2024-07-05:
I just grabbed all the .deb from the latest build at the ubuntu-asahi-next and installed them, I can confirm sound on the speaker is working, including decreasing or increasing the volume.

Thank you a lot for the help and quick fix!

tobhe, 2024-07-05:
Thank you! I couldn't have figured it out without your help :)

ansfidine, 2024-07-05:
Wow happy to see that this issue is now resolved I may try again as well with this solution if I hit the same issue again. Thanks 🙏 🎊🎊🎊🎊🎊

closed 2024-07-13 by tobhe