UbuntuAsahi/ubuntu-asahi #126: Boot LIVE USB Linux Ubuntu or Asahi direct from USB

jittygitty, 2024-06-29:
Is it possible to boot AsahiLinux or UbuntuLinux direct from USB as a LIVE USB boot iso? 

Many of us I think would like to boot LIVE ISO's or full persistent Linux installs direct from USB.

We don't want to touch the internal drive, other than perhaps just to tell internal m2 "U-Boot" to load/boot from USB.

Is this possible? If so, any links on how to do it?

jaggernaut007, 2024-09-23:
Is it possible? I am trying rn

tobhe, 2024-09-23:
I ran some tests recently and built an image that might work. It is available at https://files.tobhe.de/ubuntu/ubuntu-24.04-preinstalled-desktop-arm64%2Basahi.img.

Keep in mind you will still have to go through `curl https://alx.sh | sh` and choose the m1n1+ u-boot uefi only option to get a boot environment that supports USB booting.

There unfortunately is no way to boot from USB without installing anything on your actual disk.

eindgebruiker, 2024-11-23:
@tobhe Which U-BOOT command(s) should I run to boot from USB? 
I have your pre-built image on USB and Ubuntu Asahi on my Macbook. I tried _run bootcmd_usb0_ which is mentioned in issue 26, and the instructions from David Alger, but none of them work.

devnoname120, 2025-04-01:
@eindgebruiker Upgrade your U-Boot to the current latest version: https://launchpad.net/~ubuntu-asahi/+archive/ubuntu/ubuntu-asahi-next/+files/u-boot-asahi_2024.10-1ubuntu0.1~asahi1_arm64.deb

Then run `bootmenu` instead of `bootcmd_usb0`. Your USB device should appear and you can press the return key to proceed.

eindgebruiker, 2025-08-07:
I did not see any errors. Adding rootdelay=130 did not help either.

I tried a couple of times with another, older USB stick. That started successfully every time.

JunielKatarn, 2025-09-01:
> [@eindgebruiker](https://github.com/eindgebruiker) Upgrade your U-Boot to the current latest version: [...]

Tried installing this from Ubuntu Asahi 24.04 but u-boot does not have either `bootmenu` or `bootcmd_usb0` defined.

The device does recognize the USB with the live Ubuntu Asahi image, but `env print` doesn't show those values as defined.

What's your environment?

eindgebruiker, 2025-09-01:
@JunielKatarn I used the image mentioned here:

> I ran some tests recently and built an image that might work. It is available at https://files.tobhe.de/ubuntu/ubuntu-24.04-preinstalled-desktop-arm64%2Basahi.img.

JunielKatarn, 2026-09-26:
> ```
> snap install --classic ubuntu-image
> git clone https://github.com/UbuntuAsahi/ubuntu-asahi
> cd ubuntu-asahi
> sudo ubuntu-image classic -vv asahi-desktop.yaml
> ```

I tried this and obtained a `tar.gz` file prefixed `ubuntu-26.04`, but the image build process clearly installed GNOME 46 (corresponding to 24.04?).
Moreover, the resulting `tar.gz` is only about 4GB where the published IMG files are between 8 and 10GB.
Did I miss any customizations?

None of the published IMG (24.04, 24.10, 25.10) are correctly booting on a MacBook Pro (M1 Pro) using `bootflow -b usb`.
Any hints?