UbuntuAsahi/ubuntu-asahi #127: Only side speakers are being used on MacBook M2 Pro

tacolegs2004, 2024-06-29:
Whenever I play a video,it only uses the side speakers and ends up not sounding great.

tacolegs2004, 2024-06-29:
Also I can only turn the volume down to 95%.

tobhe, 2024-06-29:
hi @tacolegs2004! Which machine model are you using? can you try updating your packages and rebooting and see if that fixes the speakers? I would love to help you fix this but I will need a little more information to figure out what's wrong.

tacolegs2004, 2024-06-30:
I'm using an M2 Macbook Pro. I've upgraded my packages and rebooted a few times and I still get the issue.

tobhe, 2024-06-30:
Can you paste the output of the following command please: `apt list --installed | grep asahi`. We will have to identify which of the packages is missing or outdated on your system. I don't see the same bug on my m2 air so it might be macbook pro related.

tacolegs2004, 2024-07-02:
This is the output I got.

alsa-ucm-conf-asahi/noble,now 5-1 all [installed,automatic]
asahi-audio/noble,noble,now 1.8-1~bpo24.04.0~asahi0 all [installed,automatic]
asahi-configs/noble,noble,now 1-2 arm64 [installed,automatic]
asahi-scripts/noble,noble,now 20240623-1ubuntu0.1~asahi1 all [installed,automatic]
libegl-mesa0/noble,noble,now 24.1.0~pre20240228-1asahi2.24.04 arm64 [installed,automatic]
libgbm1/noble,noble,now 24.1.0~pre20240228-1asahi2.24.04 arm64 [installed,automatic]
libgl1-mesa-dri/noble,noble,now 24.1.0~pre20240228-1asahi2.24.04 arm64 [installed,automatic]
libglapi-mesa/noble,noble,now 24.1.0~pre20240228-1asahi2.24.04 arm64 [installed,automatic]
libglx-mesa0/noble,noble,now 24.1.0~pre20240228-1asahi2.24.04 arm64 [installed,automatic]
linux-asahi-arm-headers-6.8.0-1011/noble,noble,now 6.8.0-1011.11 all [installed,automatic]
linux-asahi-arm/noble,noble,now 6.8.0.1011.11 arm64 [installed,automatic]
linux-headers-6.8.0-1011-asahi-arm/noble,noble,now 6.8.0-1011.11 arm64 [installed,automatic]
linux-headers-asahi-arm/noble,noble,now 6.8.0.1011.11 arm64 [installed,automatic]
linux-image-6.8.0-1011-asahi-arm/noble,noble,now 6.8.0-1011.11 arm64 [installed,automatic]
linux-image-asahi-arm/noble,noble,now 6.8.0.1011.11 arm64 [installed,automatic]
linux-modules-6.8.0-1011-asahi-arm/noble,noble,now 6.8.0-1011.11 arm64 [installed,automatic]
m1n1/noble,noble,now 1.4.14-1ubuntu0.1~asahi1 arm64 [installed,automatic]
mesa-vulkan-drivers/noble,noble,now 24.1.0~pre20240228-1asahi2.24.04 arm64 [installed,automatic]
u-boot-asahi/noble,noble,now 2024.04-1ubuntu0.1~asahi2 arm64 [installed,automatic]
ubuntu-asahi/noble,noble,now 1.21+24.04 arm64 [installed]

tobhe, 2024-07-02:
Ok, so that looks up-to-date. Nothing obviously wrong on your side then as far as I can tell.
I also can't find any obvious related upstream bugs.
It might be worth looking at `dmesg` and `journalctl -u speakersafetyd` to see if anything suspicious shows up in the logs.

tobhe, 2024-07-02:
on second looks it might be related to https://github.com/AsahiLinux/asahi-audio/issues/41
I think I have an idea what might be wrong.

chadmed, 2024-07-13:
asahi-audio has not been installed properly, or there is a version mismatch between asahi-audio and wireplumber. The DSP profile is not being picked up.

tobhe, 2024-07-13:
@chadmed I THINK it was actually an issue with pipewire which accidentally got downgraded to a version not supporting LV2 plugins that should be fixed now.

Something went wrong with our repo pinning and our patched package was replaced with a newer one from the regular ubuntu archive. I believe this should be fixed now.

amirsalarsafaei, 2024-09-19:
I installed today and got the same issue

tobhe, 2024-09-19:
@amirsalarsafaei have you tried updating and rebooting after the update after installation? I have new installer images in the pipline which will get published once we are sure they work properly, those should properly fix the issue for new installations.