UbuntuAsahi/ubuntu-asahi #128: Ubuntu 22.04 Server installation

dbruski1, 2024-07-31:
Hi!

Is it possible from your side to update installation for Ubuntu Server 22.04?
I've tried to install it since its present on the choice list ( unsupported) while using the Asahi installer.

Although my installation fails with exception "You system firmware is too old".
[ubuntu_installation.txt](https://github.com/user-attachments/files/16439820/ubuntu_installation.txt)

SYSTEM INFORMATION:
```
  Product name: Mac mini (2023)
  SoC: Apple M2 Pro
  Device class: j474sap
  Product type: Mac14,12
  Board ID: 0x2
  Chip ID: 0x6020
  System firmware: iBoot-10151.121.1
  Boot UUID: xxx
  Boot VGID: xxx
  Default boot: xxx
  Boot mode: macOS
  OS version: 14.5 (23F79)
  OS restore version: 23.6.79.0.0,0
  Main firmware version: 14.5 (23F79)
  No Fallback System Firmware / rOS
  SFR version: 23.6.79.0.0,0
  SystemRecovery version: 23.4.56.0.0,0 (14.3 23D56)
```

tobhe, 2024-08-02:
Thanks for the report, I think the problem here is that your machine is simply too new  to work with our 22.04 server images. I would recommend waiting for 24.04 server images or installing desktop and stripping it down to what you need.

tony2019666, 2024-09-12:
same issue on m2 Mac mini. 
im trying install 22.04 server since Plesk didn't support 24.04 arm

ccgarant, 2024-11-14:
Same issue here as well. Mac mini M2. Oddly I had install errors, but then the username and password didn't work. Tried it twice made sure the password was good.

tobhe, 2024-11-14:
@ccgarant user + password on this old image is ubuntu:ubuntu by default

ccgarant, 2024-11-14:
You're kidding me 🤦‍♂️. Some kind of prompt would be helpful to know that, otherwise how?

tobhe, 2024-11-14:
We used to have this info in our README until we dropped 22.04 and enabled interactive user creation in our newer images. We then brought back 22.04 after a while because people kept on asking for it.
We probably simply forgot to also restore the old installation instructions.

tony2019666, 2024-11-14:
> Same issue here as well. Mac mini M2. Oddly I had install errors, but then the username and password didn't work. Tried it twice made sure the password was good.

May I ask how did you bypass the “You system firmware is too old" issue on M2?

ccgarant, 2024-11-14:
Sorry, glossed over that. Never was successful on the M2 with v22 (v24 desktop works). 

I also have an M1 that was running Asahi Archlinux. Tried v22 on that and was successful. But I could not login with the username password combo I thought was mine. Apparently it's ubuntu ubuntu which I did not see anywhere. I would guess most people will have the stonewall.

ccgarant, 2024-11-18:
Had trouble bringing wifi or eth up despite trying netplan, wpa_supplicant, trying to bring it UP, and hardline connecting and rebooting. Giving up.