UbuntuAsahi/ubuntu-asahi #130: Webcam not working

Kruemelmann, 2024-08-26:
Hello, 

I installed Ubuntu again today on my Macbook Pro M1 Max. My camera was and is still not working on Ubuntu 24. Do you have any ideas on this or is this known because the support of asahi is not ported yet?

many thanks for the wonderful project, by the way

tobhe, 2024-08-27:
Hi @Kruemelmann! The camera is expected to work, I just noticed that there seems to be a bug with our pre-installed camera app though.
Have you tried using it from your browser? Can you see a `/dev/video0` device on your machine?

Kruemelmann, 2024-08-30:
Yes, you are right, inside the browser the camera is working fine. In the shipped camera app it does not show up.

tobhe, 2024-08-30:
Thanks for verifying that! I'll see if I can figure out what's wrong with the built-in camera app.