UbuntuAsahi/ubuntu-asahi #134: Who is Tobias Heider and why is REPO_BASE using a personal site? 😁

RandomInsano, 2024-09-22:
I assume @tobhe is a leader on this project, but it didn't make me very confident to see a personal website used in the install script for a project like this. I'm not sure if I want to build everything from scratch right now to verify integrity 🙂. I do think having https://files.ubuntuasahi.org DNS pointing to the same server would at least allow a teammate to move to a new machine if unfortnate events happen.

Consider this a request to make the installer point to a more "official" DNS name.

tobhe, 2024-09-23:
Hey @RandomInsano, that would in fact be me. If you are irritated by my non-human looking profile pic, maybe this [video of a talk](https://www.youtube.com/watch?v=EJ8hdpXkkMI) I gave together with marcan at last year's Ubuntu Summit helps convince you that I am in fact real (and human shaped). I think some of the things you wrote require clarification so here is a longer than usual reply hopefully providing just that.

First of all, I am assuming you are talking about [this](https://github.com/UbuntuAsahi/ubuntuasahi.github.io/blob/main/static/install) script which is used during the _Ubuntu Asahi_ installation process to point the _Asahi Linux_ installer to our pre-built images currently hosted at https://files3.tobhe.de/ubuntu/os/.

> it didn't make me very confident to see a personal website used in the `REPO_BASE` variable in the install script for a project like this

Looking at the landing page of [tobhe.de](https://tobhe.de), I think is quite obvious that this is my personal domain. It even links back to my github profile. The server hosting those images is also administered by me and paid out of my own pocket. This is a remnant from when I started the project. Initially the repo was also hosted on https://github.com/tobhe/ubuntu-asahi and it didn't even have a proper website. When the project gained traction and more contributors joined the team, we moved our code to a separate github organization and another contributor registered a domain and built what I think is a really nice looking website (hosted on their own web space) at [ubuntuasahi.org](https://ubuntuasahi.org) and here we are today.

> I'm not sure if I want to build everything from scratch right now to verify integrity 🙂.

I think you highlight a very important issue here that people often overlook in open source software:  If you want to use this project you probably have to somehow make sure you can trust it. At the same time "open source is people!", we are not a corporate entity building a product, just a group of friends working on this for fun in our free time. Trusting the project means trusting *me* and *every one of us*. I don't really see how I could change this. It isn't even just the image hosting, I also build those images on my computer, running a script I wrote, installing packages I uploaded to a Launchpad PPA I have admin access to. Even worse, you also have to trust my judgement in other people, because I invited others and gave them access to the PPA, the github organization and the website. I can assure you that I picked people I trust, but that probably doesn't help if you don't trust me in the first place. That being said we would love if more people read our code, find bugs and report or fix them 😉

> I do think having https://files.ubuntuasahi.org/ DNS pointing to the same server would at least allow a teammate to move to a new machine if unfortnate events happen.

Not sure which unfortunate event you had in mind, but should I get bus-factored any other contributor with access to github or the web server can change the script to point to a different `REPO_BASE`.

> Consider this a request to make the installer point to a more "official" DNS name.

I honestly don't think it makes a difference other than hiding who is actually running the infrastructure. If you look closely enough you could still figure out that the same server also hosts my personal domain. Giving users the illusion that we aren't actual people but an abstract entity seems pointless and probably not worth spending any of my free time on. I might change it in the future but it certainly isn't far up on my todo list. I am open to reconsider this if I hear a compelling argument of course.

Now I have barely touched the surface of the trust problem with running random software from the internet but there is one more thing: We also strongly rely on other people's code (some of which none of us ever even met in person). Most prominently we ship source code and binaries from the Asahi Linux project and of course Ubuntu and by extension Debian. If you look for example at the `INSTALLER_BASE` variable you can see that that points to https://cdn.asahilinux.org/installer to download their build of https://github.com/asahilinux/asahi-installer, so we would all better trust whoever is hosting that too (If you would like to help them pay the hosting bill and support their work, please consider donating at https://asahilinux.org/support/). I do, so I decided it is fine to use their binaries instead of setting up our own build infrastructure and hosting. I like to think that they also trust us just a little because they link to our site at https://github.com/AsahiLinux/docs/wiki/SW%3AAlternative-Distros and sometimes merge our pull requests in their repos. I am also pretty sure that none of us reviewed all of the source code any of the other projects distribute.

I hope that helps you decide whether this project is for you or not. We are distributing it in the hope that it will be useful to some people, but it comes without any warranty whatsoever so I can't guarantee it is trustworthy enough for your use-case.

RandomInsano, 2024-09-27:
Thanks for the detailed write-up @tobhe! I submitted this too late at night and didn't filter for how it might be received so sorry for my tone before.

I think my main concern is that for myself personally, trust is built from many eyes reviewing each other's work. Something akin to the [Web of Tust](https://en.wikipedia.org/wiki/Web_of_trust) from PGP land. The XZ backdoor was an example where the issue wasn't in source control (so fewer reviewers) and it was detected by a [diligent engineer](https://www.openwall.com/lists/oss-security/2024/03/29/4). One one person cannot review all code, so more open and shared the build and distribution infrastructure is the more I trust I have.

The [bus factor](https://en.wikipedia.org/wiki/Bus_factor) is definitely the other bit of concern.

I can appreciate that my personal views on trust aren't a focus for the project. I think that if one day this project can be affiliated and shared with Canonical I'll give it a try! I hope the best for you folks and your users!

closed 2024-09-27 by RandomInsano