UbuntuAsahi/ubuntu-asahi #135: OpenCL Support in Ubuntu Asahi?

Cimplex, 2024-10-04:
I noticed in two different locations that people are getting their OpenCL through Asahi's ICL:

Source: https://www.reddit.com/r/AsahiLinux/comments/1fkog1a/opencl_support/

I've setting the variable: `RUSTICL_ENABLE=asahi`
And running: `clinfo`

But I still get no devices. Does anyone know if this is an issue that Ubuntu version won't support or am I missing an update, package, or something?

tobhe, 2024-10-04:
We might have to enable rusticl for our mesa builds, it looks like it currently is built without, see https://github.com/UbuntuAsahi/mesa/blob/ubuntu/noble/debian/rules#L46C1-L46C83

This could be relatively simple or harder depending on the required dependency changes, I'll let you know once we have something to test.

Cimplex, 2024-10-04:
I went digging some more and found a guy that enabled RUSTICL for fedora: https://nullr0ute.com/2023/12/getting-started-with-opencl-using-mesa-rusticl/

I can see build options in the Mesa driver itself: https://salsa.debian.org/xorg-team/lib/mesa/-/blob/debian-unstable/meson_options.txt?ref_type=heads

I do apologize if none of this is relevant or helpful. I don’t ever get out of userland, and this close to the kernel is outside my comfort zone.

Thank you for taking the time to look into this!

Cimplex, 2024-10-11:
You've probably already seen, but for everyone, asahi linux dropped this beast of a blog:

https://asahilinux.org/2024/10/aaa-gaming-on-asahi-linux/

tobhe, 2024-10-11:
Hey @Cimplex, mesa updates are currently rolling out [here](https://launchpad.net/~ubuntu-asahi/+archive/ubuntu/ubuntu-asahi/+packages) and should be available in a bit :) Let me know if that works for you once you got a chance to test it.

Cimplex, 2024-10-12:
Awesome! `clinfo` is working for me now. I'm going to move over my OpenCL app to continue testing.

I had to specifically download your `mesa-opencl-icd_24.2.0~pre20241006-1asahi2.24.04_arm64.deb` package because it wasn't included with just a normal apt update/upgrade:

My `/var/log/apt/history.log`:
```
Start-Date: 2024-10-12  03:50:48
Commandline: apt upgrade
Requested-By: ubuntu (1000)
Install: libllvm18:arm64 (1:18.1.3-1ubuntu1, automatic)
Upgrade: libglx-mesa0:arm64 (24.2.0~pre20240725-1asahi1.24.04, 24.2.0~pre20241006-1asahi2.24.04), libgbm1:arm64 (24.2.0~pre20240725-1asahi1.24.04, 24.2.0~pre20241006-1asahi2.24.04), m1n1:arm64 (1.4.15-1ubuntu0.1~asahi1, 1.4.16-1ubuntu0.1~asahi1), libgl1-mesa-dri:arm64 (24.2.0~pre20240725-1asahi1.24.04, 24.2.0~pre20241006-1asahi2.24.04), mesa-vulkan-drivers:arm64 (24.2.0~pre20240725-1asahi1.24.04, 24.2.0~pre20241006-1asahi2.24.04), libglapi-mesa:arm64 (24.2.0~pre20240725-1asahi1.24.04, 24.2.0~pre20241006-1asahi2.24.04), libegl-mesa0:arm64 (24.2.0~pre20240725-1asahi1.24.04, 24.2.0~pre20241006-1asahi2.24.04)
End-Date: 2024-10-12  03:50:49

Start-Date: 2024-10-12  03:56:11
Commandline: apt install ./mesa-opencl-icd_24.2.0~pre20241006-1asahi2.24.04_arm64.deb
Requested-By: ubuntu (1000)
Install: mesa-opencl-icd:arm64 (24.2.0~pre20241006-1asahi2.24.04), libclc-18:arm64 (1:18.1.3-1ubuntu1, automatic), libclang-rt-18-dev:arm64 (1:18.1.3-1ubuntu1, automatic), libclang-cpp18:arm64 (1:18.1.3-1ubuntu1, automatic), libclc-18-dev:arm64 (1:18.1.3-1ubuntu1, automatic), libclang-common-18-dev:arm64 (1:18.1.3-1ubuntu1, automatic)
End-Date: 2024-10-12  03:56:12

```

tobhe, 2024-10-12:
I'd say it being not getting pulled in automatically is normal unless you had it installed as a dependency for something. Thank you for testing, feel free to close this if everything works!

Cimplex, 2024-10-13:
Much appreciate your responsiveness and notification! Thank you and the rest of the Ubuntu Asahi team!

closed 2024-10-13 by Cimplex