UbuntuAsahi/ubuntu-asahi #136: Failed to boot on Mac Mini (24GB RAM, 512GB SSD)

KhiemNguyenT, 2024-10-09:
I'm following the step-by-step instructions to install Asahi Ubuntu to my Mac mini (24GB RAM, 512GB SSD) but it failed to boot to Ubuntu.

My configuration:
 1. Reserve 30% for Mac and 70% for Ubuntu
 2. Run the script "curl -sL https://ubuntuasahi.org/install | bash" and select either "Ubuntu 23.10" or "Ubuntu 24.04"
 3. Follow the guidance in the scripts
 
 In the end, when booting up to Ubuntu, I select "Ubuntu" and wait. Just see the Ubuntu logo in the middle of the screen. Nothing happens after 15 mins waiting.
 
 NOTE:
 1. I can switch back to MacOS by rebooting and select MacOS as boot partition.
 2. When Ubuntu failed to boot, I switch back to macOS, delete the Ubuntu partitions, and try again with another Ubuntu version. Not yet successful to boot any Ubuntu versions.

tobhe, 2024-10-09:
Thanks for the report, I am not aware of any issues so we will have to figure out what is happening here. Is this an M1 or M2 mini? MacOS version? Can you hit escape after selecting Ubuntu, that should give you a boot log instead of only the Ubuntu logo. Maybe there are error messages that would tell us what's wrong.

KhiemNguyenT, 2024-10-09:
Mine is M2 mini. MacOS version v14.7.
I'm in the way to reinstall another Ubuntu version. Will report back the boot logs in next hours.

tobhe, 2024-10-09:
Good to know, thanks! You might have hit an unlucky bug that is not easy to reproduce then. Let me know if any more issues show up!

KhiemNguyenT, 2024-10-09:
I tried UbuntuAsashi with Ubuntu 22.04 LTS but it failed to start up. I could not see the Ubuntu logo either. It seems there is boot - reset(panic) cycle in the background.

Since that version is about 2 years ago, it did not support M2, right ?

KhiemNguyenT, 2024-10-14:
Finally, I decided to install Ubuntu 24.04 version.

closed 2024-10-14 by KhiemNguyenT