UbuntuAsahi/ubuntu-asahi #137: TouchBar functioning as touchscreen.

akosovari, 2024-10-15:
The touchbar is recognized as a touchscreen, even after installing tiny-dfr for touchbar support. It works, but on every click it's also recognized as a touch on a screen. Quite annoying. I was experimenting with methods to block this behavior but didn't get any usable results. Any ideas how to solve?

tobhe, 2024-10-15:
can you paste your kernel dmesg? I fell like this should be solvable by blocking the wrong module to load

akosovari, 2024-10-16:
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x611f0221]
[    0.000000] Linux version 6.11.0-1001-asahi-arm (buildd@bos03-arm64-045) (aarch64-linux-gnu-gcc-14 (Ubuntu 14.2.0-4ubuntu2) 14.2.0, GNU ld (GNU Binutils for Ubuntu) 2.43.1) #1-Ubuntu SMP PREEMPT_DYNAMIC Sat Sep 28 14:42:09 UTC 2024 (Ubuntu 6.11.0-1001.1-asahi-arm 6.11.0)
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Apple MacBook Pro (13-inch, M1, 2020)
[    0.000000] efi: EFI v2.10 by Das U-Boot
[    0.000000] efi: RTPROP=0xbda3db040 INITRD=0xbda3b0040 MEMRESERVE=0xbda3a7040 
[    0.000000] secureboot: Secure boot disabled
[    0.000000] OF: reserved mem: 0x0000000800b18000..0x000000080153bfff (10384 KiB) nomap non-reusable asc-firmware@800b18000
[    0.000000] OF: reserved mem: 0x0000000801b5c000..0x0000000801f8bfff (4288 KiB) nomap non-reusable asc-firmware@801b5c000
[    0.000000] OF: reserved mem: 0x0000000805be8000..0x0000000805c47fff (384 KiB) map non-reusable flash@805be8000
[    0.000000] OF: reserved mem: 0x0000000bdf530000..0x0000000bdf533fff (16 KiB) map non-reusable flash@bdf530000
[    0.000000] OF: reserved mem: 0x0000000bdf534000..0x0000000bdf6dffff (1712 KiB) nomap non-reusable isp-heap
[    0.000000] OF: reserved mem: 0x0000000be2590000..0x0000000be52ebfff (46448 KiB) nomap non-reusable framebuffer@be2590000
[    0.000000] OF: reserved mem: 0x0000000be52ec000..0x0000000be53abfff (768 KiB) nomap non-reusable region95@be52ec000
[    0.000000] OF: reserved mem: 0x0000000be53ac000..0x0000000be53affff (16 KiB) nomap non-reusable region94@be53ac000
[    0.000000] OF: reserved mem: 0x0000000be53b0000..0x0000000be63affff (16384 KiB) nomap non-reusable region57@be53b0000
[    0.000000] OF: reserved mem: 0x0000000be63b0000..0x0000000be66affff (3072 KiB) nomap non-reusable dcp_data@be63b0000
[    0.000000] OF: reserved mem: 0x0000000bfff70000..0x0000000bfff73fff (16 KiB) nomap non-reusable uat-handoff
[    0.000000] OF: reserved mem: 0x0000000bfff78000..0x0000000bfffb7fff (256 KiB) nomap non-reusable uat-pagetables
[    0.000000] OF: reserved mem: 0x0000000bfffb8000..0x0000000bfffbbfff (16 KiB) nomap non-reusable uat-ttbs
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000800b18000-0x0000000bfffbbfff]
[    0.000000] NUMA: NODE_DATA [mem 0xbdece6980-0xbdecebfff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000800b18000-0x0000000bfffbbfff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000]   Device   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000800b18000-0x000000080153bfff]
[    0.000000]   node   0: [mem 0x0000000801b5c000-0x0000000801f8bfff]
[    0.000000]   node   0: [mem 0x0000000802084000-0x0000000805c9ffff]
[    0.000000]   node   0: [mem 0x0000000805ca0000-0x0000000805d7bfff]
[    0.000000]   node   0: [mem 0x0000000805d7c000-0x000000080799bfff]
[    0.000000]   node   0: [mem 0x000000080799c000-0x00000008079bbfff]
[    0.000000]   node   0: [mem 0x00000008079bc000-0x0000000bda3d7fff]
[    0.000000]   node   0: [mem 0x0000000bda3d8000-0x0000000bda3fffff]
[    0.000000]   node   0: [mem 0x0000000bda400000-0x0000000bdf43ffff]
[    0.000000]   node   0: [mem 0x0000000bdf440000-0x0000000bdf44ffff]
[    0.000000]   node   0: [mem 0x0000000bdf450000-0x0000000bdf52bfff]
[    0.000000]   node   0: [mem 0x0000000bdf530000-0x0000000bdf533fff]
[    0.000000]   node   0: [mem 0x0000000bdf534000-0x0000000bdf6dffff]
[    0.000000]   node   0: [mem 0x0000000be2590000-0x0000000be66affff]
[    0.000000]   node   0: [mem 0x0000000bfff70000-0x0000000bfff73fff]
[    0.000000]   node   0: [mem 0x0000000bfff78000-0x0000000bfffbbfff]
[    0.000000] Initmem setup node 0 [mem 0x0000000800b18000-0x0000000bfffbbfff]
[    0.000000] On node 0, zone DMA: 710 pages in unavailable ranges
[    0.000000] On node 0, zone DMA: 392 pages in unavailable ranges
[    0.000000] On node 0, zone DMA: 62 pages in unavailable ranges
[    0.000000] On node 0, zone DMA: 1 pages in unavailable ranges
[    0.000000] On node 0, zone DMA: 2988 pages in unavailable ranges
[    0.000000] On node 0, zone DMA: 9776 pages in unavailable ranges
[    0.000000] On node 0, zone DMA: 1 pages in unavailable ranges
[    0.000000] On node 0, zone DMA: 17 pages in unavailable ranges
[    0.000000] cma: Reserved 32 MiB at 0x0000000808000000 on node -1
[    0.000000] percpu: Embedded 20 pages/cpu s202328 r8192 d117160 u327680
[    0.000000] pcpu-alloc: s202328 r8192 d117160 u327680 alloc=20*16384
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5 [0] 6 [0] 7 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ACTLR virtualization (IMPDEF, Apple)
[    0.000000] CPU features: detected: Address authentication (IMP DEF algorithm)
[    0.000000] CPU features: detected: TSO memory model (Apple)
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: BOOT_IMAGE=/vmlinuz-6.11.0-1001-asahi-arm root=UUID=3bbe1eef-4343-4fa8-938c-a2ff577c202a ro quiet splash vt.handoff=7
[    0.000000] Unknown kernel command line parameters "splash BOOT_IMAGE=/vmlinuz-6.11.0-1001-asahi-arm", will be passed to user space.
[... cut ...]
[    4.989435] apple-dcp 231c00000.dcp: RTKit: syslog message: PPipeDCP_H13P.cpp:18011: Dual pipe disabled offset 0
[    4.989461] apple-dcp 231c00000.dcp: RTKit: syslog message: nifiedPipeline.cpp:7465: set_digital_out_mode returned 8000000b
[    4.989465] apple-dcp 231c00000.dcp: set_digital_out_mode finished:8500
[   19.986592] systemd-journald[402]: /var/log/journal/8dc429a15f9b43b49cc13b1071948330/user-1000.journal: Journal file uses a different sequence number ID, rotating.
[   20.030573] kauditd_printk_skb: 156 callbacks suppressed