UbuntuAsahi/ubuntu-asahi #142: Speaker volume stuck at 100%, does not allow change (M1 Max / 16" Macbook Pro)

madcow4242, 2024-10-29:
Volume control works fine at boot time, and even after multiple suspend/resume cycles.  However, at as-yet-unknown intervals the volume control stops working correctly.

Trying to change volume with the applicable function keys brings on the OSD for volume, but the displayed volume setting only will toggle between full volume and one setting down from full (no matter what key is pressed).

Similarly, once this starts to happen, changing volume in the Settings/Sound control panel also does not function (it stays at full volume setting).

A full reboot resets this issue until some future time when it will recur.  If I can figure out what circumstances trigger it I'll add context.

tobhe, 2024-10-30:
Thanks for the report, this seems to be a bug in gnome related to our arguably rather complex audio setup. There are audio devices exposed to userland which speakersafetyd is supposed to attach to and then the real safe device that gnome is actually supposed to use. Every once in a while it seems to pick up the wrong one (I think that might be after suspend).

madcow4242, 2024-10-30:
I'll keep an eye on it but if it's after suspend, it's not ALWAYS after suspend.  I can suspend/resume quite a number of times with no issue, then I'll find it's happened after that.  I'll ensure to test the speaker volume control immediately before/after each suspend and see if it triggers at that specific event.

madcow4242, 2024-10-30:
I just had it occur, and it wasn't due to suspend/resume - it happened immediately after a system notification / beep - in this instance, a pop-up notice at the top of the screen from an incoming Mattermost direct message.  After that "beep", volume is stuck at 100%.   (I would screen shot the message, but it disappears when I use the tool)

hamza72x, 2024-12-27:
+1 Same Here

tobhe, 2024-12-28:
You can temporarily work around this using `systemctl --user restart pipewire`. Still looking for a permanent fix.