UbuntuAsahi/ubuntu-asahi #143: Issues Installing amd64 and i386 Packages on ARM64 Ubuntu (MacBook Air M1)

kvkvlaa, 2024-10-29:
I'm running Ubuntu 24.04.1 LTS on a MacBook Air M1, which uses an ARM64 architecture. I'm facing a few issues trying to install packages that require the amd64 and i386 architectures.

Here's what I've tried so far:

    Wine and Steam Installation: When trying to install Wine and Steam, I've encountered dependency issues where certain libraries (like libc6:i386 and libegl1-mesa:arm64) either have "no installation candidate" or simply can't be found.

    Enabling Multi-Architecture Support: I've tried enabling amd64 and i386 architectures with:
sudo dpkg --add-architecture i386
sudo dpkg --add-architecture amd64
sudo apt update

However, I still receive 404 errors when running apt update, specifically pointing out missing i386 packages, as well as unresolved dependencies for amd64 packages.
Flatpak Installation: I attempted installing Steam via Flatpak as an alternative, but the Steam package doesn’t seem to be available on Flathub for ARM64, and installing through Flatpak hasn't resolved the compatibility issues.

My main goal is to run amd64 (and possibly i386) applications, particularly Steam and Wine, on ARM64 Ubuntu. Does anyone have a workaround, or has anyone successfully set up something similar on an ARM64 Ubuntu system?

tobhe, 2024-10-30:
In short: you can't. The only way to run amd64 apps is through emulation, we don't have an easy way to set up emulation yet. On the Fedora side they worked on a steam package that does all the setup for you, this is still WIP for us.

FelicianoTech, 2024-12-09:
@tobhe Where is that work (a Steam package like Fedora) happening and how can I keep tabs on it?

ccgarant, 2024-12-09:
The ArchLinux Asahi has a Fedora remix, maybe this will help?

https://asahilinux.org/

kaazoo, 2024-12-09:
Hi, I'm working on packaging of Steam and dependencies for Ubuntu 24.04 and 24.10.
See https://kernelpanik.net/vulkan_and_steam_on_ubuntu_asahi/ for a work in progress howto.