UbuntuAsahi/ubuntu-asahi #144: Hardware acceleration not working on Mac mini M1

kirb, 2024-11-03:
I installed Ubuntu Asahi 24.04 on a clean DFU-restored Mac mini M1, and ended up with software rendering:

[image]

The Gnome desktop under Wayland is laggy as expected for software rendering.

Firefox about:support shows that the Asahi mesa build is being used, but is still rendering with llvmpipe.

[image]

Any idea what's going on here?

Here's my `apt list --installed`: [list.txt](https://github.com/user-attachments/files/17610060/list.txt)

When `update-initramfs` runs, these lines are printed. Doubt it's related, just mentioning it in case.

```
Processing triggers for linux-image-6.11.0-1001-asahi-arm (6.11.0-1001.1) ...
/etc/kernel/postinst.d/initramfs-tools:
update-initramfs: Generating /boot/initrd.img-6.11.0-1001-asahi-arm
W: Possible missing firmware /lib/firmware/asmedia/asm2214a-apple.bin for built-in driver xhci_pci
W: Possible missing firmware /lib/firmware/renesas_usb_fw.mem for built-in driver xhci_pci
W: Possible missing firmware /lib/firmware/renesas_usb_fw.mem for built-in driver xhci_pci
```

tobhe, 2024-11-03:
That is unexpected. It looks like your kernel version is ok but mesa seems to be outdated. The latest mesa is `24.3.0~pre20241006-1asahi3.24.04`. Does an `apt update` followed by `apt upgrade` fix the issue?

kirb, 2024-11-03:
Sorry, forgot to specify, everything is up to date.

<details>

```
❯ sudo apt update                    
Hit:1 https://deb.nodesource.com/node_20.x nodistro InRelease
Hit:2 https://download.docker.com/linux/ubuntu noble InRelease                                                                                                                  
Hit:3 https://packages.microsoft.com/repos/code stable InRelease                                                                                                                
Hit:4 http://ports.ubuntu.com/ubuntu-ports noble InRelease                                                                                                
Hit:5 http://ports.ubuntu.com/ubuntu-ports noble-updates InRelease                                                                  
Get:6 https://pkgs.tailscale.com/stable/ubuntu noble InRelease                                         
Hit:7 http://ports.ubuntu.com/ubuntu-ports noble-backports InRelease
Hit:8 https://ppa.launchpadcontent.net/ubuntu-asahi/ubuntu-asahi/ubuntu noble InRelease
Hit:9 http://ports.ubuntu.com/ubuntu-ports noble-security InRelease
Fetched 6578 B in 2s (3847 B/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.

❯ sudo apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```
</details>

Package versions:

```
❯ apt list --installed | grep mesa
libegl-mesa0/noble,now 24.3.0~pre20241006-1asahi3.24.04 arm64 [installed,automatic]
libgl1-mesa-dri/noble,now 24.3.0~pre20241006-1asahi3.24.04 arm64 [installed,automatic]
libglapi-mesa/noble,now 24.3.0~pre20241006-1asahi3.24.04 arm64 [installed,automatic]
libglu1-mesa/noble,now 9.0.2-1.1build1 arm64 [installed,automatic]
libglx-mesa0/noble,now 24.3.0~pre20241006-1asahi3.24.04 arm64 [installed,automatic]
mesa-libgallium/noble,now 24.3.0~pre20241006-1asahi3.24.04 arm64 [installed,automatic]
mesa-utils-bin/noble,now 9.0.0-2 arm64 [installed,automatic]
mesa-utils/noble,now 9.0.0-2 arm64 [installed,automatic]
mesa-vulkan-drivers/noble,now 24.3.0~pre20241006-1asahi3.24.04 arm64 [installed,automatic]

❯ apt list --installed | grep linux
binutils-aarch64-linux-gnu/noble-updates,noble-security,now 2.42-4ubuntu2.3 arm64 [installed,automatic]
console-setup-linux/noble,now 1.226ubuntu1 all [installed,automatic]
cpp-13-aarch64-linux-gnu/noble,now 13.2.0-23ubuntu4 arm64 [installed,automatic]
cpp-aarch64-linux-gnu/noble,now 4:13.2.0-7ubuntu1 arm64 [installed,automatic]
libselinux1/noble,now 3.5-2ubuntu2 arm64 [installed,automatic]
linux-asahi-arm-headers-6.11.0-1001/noble,now 6.11.0-1001.1 all [installed,automatic]
linux-asahi-arm/noble,now 6.11.0.1001.2 arm64 [installed,automatic]
linux-base/noble,now 4.5ubuntu9 all [installed,automatic]
linux-firmware/noble-updates,now 20240318.git3b128b60-0ubuntu2.4 arm64 [installed,automatic]
linux-headers-6.11.0-1001-asahi-arm/noble,now 6.11.0-1001.1 arm64 [installed,automatic]
linux-headers-asahi-arm/noble,now 6.11.0.1001.2 arm64 [installed,automatic]
linux-image-6.11.0-1001-asahi-arm/noble,now 6.11.0-1001.1 arm64 [installed,automatic]
linux-image-asahi-arm/noble,now 6.11.0.1001.2 arm64 [installed,automatic]
linux-modules-6.11.0-1001-asahi-arm/noble,now 6.11.0-1001.1 arm64 [installed,automatic]
linux-sound-base/noble,now 1.0.25+dfsg-0ubuntu7 all [installed,automatic]
pptp-linux/noble,now 1.10.0-1build4 arm64 [installed,automatic]
util-linux/noble-updates,now 2.39.3-9ubuntu6.1 arm64 [installed,automatic]
```

In terms of Firefox, from `snap list mesa-2404` I see version 24.2.0\~pre20240725 from asahi/stable. I switched it to the asahi/beta version, which is 24.2.0\~pre20241006, but I'm still seeing 20240725 in about:support... I don't know what I'm missing there, need to tell snap to use the beta version for Firefox I guess?

I installed a deb of Firefox Nightly and got the 20241006 version, but it's still using llvmpipe.

[image]

tobhe, 2024-11-03:
firefox is based on core22 so it pulls mesa from `gnome-42-2204`. The version does indeed look like the latest snap version but that should not be a problem. I have the same on my machine and it works:

[image]

I am also pretty sure I tested the same image with my m1 mini and that worked. I think we need to find out why even your native mesa doesn't work, the snap probably fails for the same reason.

corrosive4354, 2024-11-04:
The firmware missing warning you mentioned may be related to the USB driver, but it is unlikely to directly cause the graphics acceleration problem. Nevertheless, you can try to install the missing firmware:
sudo apt install firmware-linux

Also, seeing that you are using Docker, I would like to recommend [Servbay](https://servbay.com) to you, which is more professional and easy to use than Docker on Mac

kirb, 2024-11-04:
linux-firmware is already installed and up to date:

```
❯ sudo apt install linux-firmware
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
linux-firmware is already the newest version (20240318.git3b128b60-0ubuntu2.4).
linux-firmware set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

Reinstalled just to be sure, and I do still get those same warnings, and still no hardware rendering. USB-A ports seem completely fine though (haven't tried USB-C).

I also reinstalled every package just to be super sure (`sudo apt install --reinstall $(apt list --installed | tail -n +2 | cut -d/ -f1)`) but again, no change. Trying an identical install on my MacBook Air M1 to see if I can reproduce the problem there.

Thanks for the hint, I've been using OrbStack on macOS, but ServBay looks a lot more complete of a solution. This Mac mini is going to be a home server and I just find macOS not good for that (that's why I installed Ubuntu Asahi), but it'll be perfect for my MacBook Pro!

tobhe, 2024-11-04:
It persist after a reinstall? Then I really don't know what the problem could be anymore. It must be a hardware specific issue because I know for sure that other installations with the same image have not had that problem.

What macos version is installed? You could try updating that to the latest and see if it makes a difference.

kirb, 2024-11-04:
I didn't reinstall yet, was actually trying it on a different M1 Mac. The result of that was that the M1 MacBook Air worked totally fine out of the box, "Apple M1 (G13G B1)". Genuinely lost about what the problem could be now 😅

The M1 Mac mini which is having the issue was DFU restored to macOS 15.1 (latest as of last week), and then all I did was run the install script in Terminal. macOS install was shrunk to the minimum the install script allowed, since I don't intend to be booted to macOS except for maintenance.

The M1 MacBook Air which is working fine has an existing install of macOS, which I updated from 15.0 to 15.1 before running the install script. I followed the exact same steps, other than only giving the Linux partition 40GB rather than 400GB. Also tried retracing my post-install steps with my [setup script](https://github.com/kirb/dotfiles/blob/main/setuplinux), just in case it actually broke later when I installed something, but it's still working fine.

Are there any logs you'd like me to pull? If not, I'll just nuke and retry the install now.

tobhe, 2024-11-06:
Probably fine to nuke it. If it happens again we can still look for logs, though at this point I don't even know what to look for.

kirb, 2024-11-22:
I just got around to doing a reinstall, GPU is correctly displaying as “Apple M1 (G13G B1)” in System Details and Firefox now.

I did notice that connecting via gnome-remote-desktop’s recently added “Remote Login” mode causes software rendering. I’m sure that’s not specific to Ubuntu Asahi though - Wayland and remote desktop are known to not exactly get along well, especially under Gnome. The issue was previously happening when logged in locally via HDMI, though, and not only via Remote Login. Still a mystery why that install broke, but at least it’s working now.