UbuntuAsahi/ubuntu-asahi #145: 24.10 Oracular Oriole support

Rad0van, 2024-11-03:
Hi,

is there any plan to support 24.10 Oracular Oriole any time soon? Particularly I'd like to install KDE Desktop with the new Plasma 6.1

Thank you.

tobhe, 2024-11-04:
Hey, you can install 24.04 and upgrade to 24.10 with `update-manager`. We are still working on 24.10 pre-built installer images but functionally there should be no difference.

FelicianoTech, 2024-12-09:
As mentioned above, you can run Ubuntu Asahi 24.10 if you upgrade from the previous release. I was running 24.04 and just upgraded about an hour ago by running `sudo do-release-upgrade`. 

I haven't been running it long, but so far, it's been good, aside from snaps like Firefox trying to take over again. That's a Canonical problem, though, not an Ubuntu Asahi problem.

closed 2024-12-09 by tobhe