UbuntuAsahi/ubuntu-asahi #147: Apple Magic Trackpad does not function with Ubuntu Asahi

madcow4242, 2024-11-07:
I can get the trackpad to pair and connect, but only the single-click and double-click functionality works.

Right click does not function
Track pad movement or scrolling does not function

Unpaired / repaired / rebooted multiple times, same results.

External Apple Magic Trackpad
M1 Max 16" Macbook Pro, 2021

tobhe, 2024-11-10:
Thanks for the report! Is this a new magic trackpad? Looks like support for the 2024 version is not in mainline yet but there is a patch on the mailing list, see https://bugzilla.kernel.org/show_bug.cgi?id=219470 and https://lore.kernel.org/lkml/20241110002816.6064-1-callahankovacs@gmail.com/

madcow4242, 2024-11-12:
I should have included the trackpad model details, but it's a older one -
not 2024 for sure.  It is model A1339

-- 
[image: Canonical-20th-anniversary]

Kevin Cazabon

Silicon Alliances Ecosystem Development Manager

Email:

***@***.***

Location:

United States (EST)

Mobile:

+1 585.301.7412

canonical.com

ubuntu.com

tobhe, 2024-11-12:
I wonder if this might be caused by https://github.com/AsahiLinux/linux/commit/88ebed4f54eb1d0a033b3ed3e9392afa8cbf49bc
The commit message questions whether this works for bt/usb devices, it might actually not.

tobhe, 2024-11-12:
Does anything interesting show up in dmesg when you pair the device?

madcow4242, 2024-11-13:
Yes, thanks for reminding me to do that.  :)

[501321.121010] magicmouse 0005:05AC:030E.0008: unknown main item tag 0x0
[501321.121098] input: kevin’s trackpad as
/devices/platform/soc/590000000.pcie/pci0000:00/0000:00:00.0/0000:01:00.1/bluetooth/hci0/hci0:12/0005:05AC:030E.0008/input/input11
[501321.121356] magicmouse 0005:05AC:030E.0008: input,hidraw5: BLUETOOTH
HID v1.60 Mouse [kevin’s trackpad] on f4:d4:88:6d:63:dc
[501321.389071] magicmouse 0005:05AC:030E.0008: unable to request touch
data (-5)
[501321.494035] magicmouse 0005:05AC:030E.0008: unable to request touch
data (-5)
[501321.558965] magicmouse 0005:05AC:030E.0008: unable to request touch
data (-5)
[501321.633996] magicmouse 0005:05AC:030E.0008: unable to request touch
data (-5)

It also says "magicmouse" although it's a magic trackpad.  I hope that
helps!

Kevin.

-- 
[image: Canonical-20th-anniversary]

Kevin Cazabon

Silicon Alliances Ecosystem Development Manager

Email:

***@***.***

Location:

United States (EST)

Mobile:

+1 585.301.7412

canonical.com

ubuntu.com