UbuntuAsahi/ubuntu-asahi #149: Add i386 build for Mesa

fathonix, 2024-11-09:
To start support for muvm and FEX on Ubuntu Asahi, an i386 build of Mesa is necessary to enable 3D acceleration on 32-bit apps such as Wine.

Here's muvm w/ sommelier running Wine64 using FEX + custom FEX RootFS w/ amd64 Ubuntu Asahi Mesa.

[image]

mkurz, 2024-12-27:
@fathonix Can you share how you achieved this setup? Thanks!

tobhe, 2024-12-28:
fwiw we now have a i386 build and some more parts of the stack are WIP. @mkurz see https://kernelpanik.net/vulkan_and_steam_on_ubuntu_asahi/ for instructions on how to set up muvm. We are working on moving all of this to our main ppa.

mkurz, 2025-01-05:
How do you generated the rootfs? Is it done via https://github.com/kaazoo/RootFS?
/cc @kaazoo
Thanks!

closed 2025-05-29 by fathonix