UbuntuAsahi/ubuntu-asahi #152: Successful V24.04 Server, Can't Login

ccgarant, 2024-11-18:
I successfully installed v24.04 server on my mac mini m1. I did this twice so I know I didn't mess up the password step.
- updated hostname prompt
- updated password prompt (repeated twice)
- localhost login
   - my login hostname and password don't work
   - ubuntu ubuntu doesn't work
   - ubuntu and new password doesn't work

Please help / advise.

tobhe, 2024-11-18:
It should boot into an interactive command line prompt to configure the root password, hostname and time zone on first boot. You should be able to log in as with username `root` and the password you provided.

ccgarant, 2024-11-18:
Thanks, I performed the interactive command line.

Looks like root username and my password worked, thanks!

Based on v22 server load and now v24, highly recommend updating the workflow to the users login, very confusing. Or add notes during the process. There's just no way to know unless you're a power user.

tobhe, 2024-11-18:
Thanks for the feedback! Yes this is a new approach we introduced with 24.04 that brings us closer to the desktop image where we also have a first boot setup instead of pre-configured passwords. The fact that the server cli solution only supports root user setup is a limitation of [systemd-firstboot](https://www.freedesktop.org/software/systemd/man/latest/systemd-firstboot.html). Ideally it would allow you to add a user too and then things would be more self explanatory.

That being said, I think 24.04 is already a huge improvement compared to an entirely pre-configured setup where it is impossible to find the credentials without looking at external docs. I can see how coming from 22.04 it might be a bit confusing, maybe we can add some text to motd or the interactive setup header to make things clearer.

ccgarant, 2024-11-18:
Nice thanks. That or a wiki faq common loggin troubleshooting. Cheers.

closed 2025-03-28 by tobhe