UbuntuAsahi/ubuntu-asahi #153: Recommended way to setup Wifi after v24.04 server successful install

ccgarant, 2024-11-18:
What is the recommended way to setup a wifi or ethernet connection after successful install? I also ran into this issue with v22 server install. If netplan in `/etc/netplan` what is the recommended filename? Also, is there a group name to set them UP? (E.g. sudo ip link set wlp1s0f0 up)

I have a mac mini m1 with `ip a` of end0 and wlp1s0f0. 

> ip a
...
> end0: <BROADCAST, MULTICAST> MTU 1500 qdisc noop state DOWN group default qlen 1000
>link/ether 00: ....
> altname enp3s0
...
> wlp1s0f0: <BROADCAST, MULTICAST> MTU 1500 qdisc noop state DOWN group default qlen 1000
...

tobhe, 2024-11-18:
The netplan docs have a [wifi example](https://netplan.readthedocs.io/en/stable/examples/#how-to-configure-your-computer-to-connect-to-your-home-wi-fi-network) that should almost work. On Ubuntu Server you need to use `renderer: networkd`. `renderer: NetworkManager` won't work because NetworkManager is not installed.

It should looks something like:
```
network:
  version: 2
  renderer: networkd
  wifis:
    wlp1s0f0:
      dhcp4: yes
      access-points:
        "network_ssid_name":
          password: "**********"
```
The file name doesn't matter as long as in ends in `.yaml` and is placed in `/etc/netplan/`, so `default.yaml` or `wifi.yaml` should both work.

ccgarant, 2024-11-19:
Agree but this is assuming you don't have any internet connections yet to install / update.

ccgarant, 2024-11-19:
Still couldn't get wifi up but at least I got ethernet end0 working from the netplan yaml...

tobhe, 2024-11-19:
@ccgarant if you run `ip link set wlp1s0f0 up` what does that return? You might need to `apt install rfkill` and `rfkill unblock all`.

ccgarant, 2024-11-19:
Thanks @tobhe, I set wlp1s0f0 to "up" and no errors. I installed rfkill and ran those commands, all good.

Now upon `sudo systemctl restart ssh` (enable and start tried before), there seems to be the following status errors.

Ssh.service not found.

I manually created one and type it into `/etc/systemd/system/ssh.service`

I've tried `sudo apt install openssh-server` and there seems to be a dependency package warning.

Upon daemon-reload, and restart ssh, there is another warning.

Ssh.socket not found. Not sure if this is all from openssh-server not installing correctly...

tobhe, 2024-11-19:
This looks like ssh socket activation: https://discourse.ubuntu.com/t/sshd-now-uses-socket-based-activation-ubuntu-22-10-and-later/30189
in short: instead of managing ssh via the service it is now managed via the ssh.socket. That means systemd starts the service automatically as soon as it sees a network packet arrive on the ssh port. Restart does not exist because a new process is spawned for each connection.

jmssa, 2025-08-20:
> Still couldn't get wifi up but at least I got ethernet end0 working from the netplan yaml...

For everyone like me who is trying to figure out why the wifi is not working after changing the .yaml file, you are probably using a WPA protected network. As such Netplan needs a package called wpasupplicant which was missing by default. Without ethernet, I believe the only way to install it is with a USB drive, be aware that you need to install both wpasupplicant and libpcsclite1 (which is a missing dependency). Hope this helps!